motioneye-project/motioneyeos #1207: Long term reliability

Chiny91, 2017-10-27:
I've just returned after 2 months away, to my 2 * RPi3 motionEyeOS systems each with 3 IPcams (6 cams).
`motionEye Version | 0.36.1
-- | --
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710` performed flawlessly, working as well today as it did for the entire 2 months, grabbing many many motion-sensed pictures and uploading them to my NAS.   Additionally, throughout the 2 months, I was able to VPN in, keeping an occasional eye on the streaming video as well as reviewing the NAS pics.   This is impressive reliability.

So, I'm all ready to break it all with something new, perhaps **the OMX h264 hardware acceleration**, when that gets released.

Great work 😄

ccrisan, 2017-10-27:
@Chiny91 thanks for the positive feedback. I don't know if you know already or not, but we have a [20171008](https://github.com/ccrisan/motioneyeos/releases/tag/20171008), admittedly a prerelease, that contains all of @jasaw's work on OMX hardware acceleration. Basically h264 movies are encoded by the hardware on a RPi. Isn't that what you're looking for? Or are you expecting a proper stable release?

Chiny91, 2017-10-28:
Ah OK, well, I usually prefer a stable release so as to minimise the number of disturbances to what is part of my security system (I know, no guarantees with motionEyeOS, RPi are educational etc).   But a daily movie would be handy, so I may make an exception, assuming no stable release due "soon" ?

Prior to departure, I'd also been tweaking my router software and every week it would email a log, part of which looked like:
```
The top 10 noisiest name clients:
 --------------------------------------------------------

 count for IP, client name: count for domain - percentage
 --------------------------------------------------------
 30104  aa.bb.cc.dd     system2:                    30106  www.google.com   100.01% 
 30104  ww.xx.yy.zz     system1:                    30107  www.google.com   100.01% 
```
Over 30k hits per week per motionEyeOS on Google, seems about right, for checking the network connection every 20 seconds.   I'm not clear whether motionEyeOS is happy with a response from my router or goes on to ping (or something) Google itself.

jasaw, 2017-10-28:
Honestly I have not tested OMX with multiple cameras. Memory bandwidth between GPU and ARM is going to be the bottleneck. May need to increase the GPU memory too. I don't know what the optimal configuration is. If you do try it out, please report back your findings.

Chiny91, 2017-11-05:
I chickened out, sorry 😞   One wet afternoon, I was tinkering around with my NAS, to which I rsync the still photos from my IPcams, and found the NAS has ffmeg.   One Python script and one cron job later, I get the videos encoded on my NAS with no RPi3 cpu/gpu stress.   I'll stick with the proven reliability of motionEyeOS as I have it running now, for now.

closed 2017-11-05 by Chiny91