motioneye-project/motioneyeos #1208: Reboots on first startup

singleflow, 2017-10-27:
I'm trying to setup a Pi Zero W, but it reboots as soon as I power it on. The Pi Zero doesn't have an ethernet port, and I cant' setup the wifi if it won't let me boot the device.

cuteboi, 2017-10-28:
You set up the wifi through the SD card.

https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

cu, 2017-11-01:
I'm not the reporter but I'm having the same problem. The Pi begins booting and then says, `ethernet: no network connection available, rebooting` and then does so.

I tried creating a `wpa_supplicant.conf` file and putting it on the `/boot` partition with no luck. Still does the same thing. Here's the contents of the file:

```
country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    ssid="eilnet2"
    psk="censored"
}
```

jasaw, 2017-11-01:
@cu `/boot/wpa_supplicant.conf` is only read once the first time device boots up. If you have booted the device with no `wpa_supplicant.conf` file or invalid `wpa_supplicant.conf` file, then you'll need to either reflash the SD card or overwrite the `/data/etc/wpa_supplicant.conf` file.

cu, 2017-11-02:
@jasaw Oh, I didn't realize that it was only read from /boot on the very first boot of a newly-flashed card. I tried that and it's working fine now. Thanks!

jasaw, 2017-11-02:
I've added this info to the wiki.

closed 2017-11-02 by jasaw