motioneye-project/motioneyeos #1209: movies not continuing eventhough there's still motion

cuddylier, 2017-10-27:
Hi

I'm having an issue where the movies don't continue even when there's motion even continuing e.g. a movie just cut off when motion was still occurring and no new ones were created after a total of three being created (3 due to my maximum length settings of 15s):

https://image.ibb.co/fprFo6/RoZ_eE_Y.png

Any ideas on what motion detection settings I might need to adjust?

I've also seen issues with large delays in motion events starting eventhough there's a massive object moving in the field of view, I tried setting 'Captured Before' to 200 frames and it helped for some motion events but equally I found some where that seemed to be ignored and the motion event didn't start until there was considerable movement for ~2+ seconds.

Thanks

cuddylier, 2017-10-27:
Just had another motion event there now, motion only ended 2 seconds before the 15 second clip was finished meaning it should have started a 2nd 15 second clip to cover the 'motion gap' setting I've got of 15 seconds.

jasaw, 2017-10-27:
Try lowering minimum motion frames and motion threshold. Captured before of 200 frames is huge, will consume a lot of RAM. Make sure you don't run out of memory.

cuddylier, 2017-10-28:
@jasaw Thanks for the suggestion, it has maybe helped a little.

At the start of motion events now for maybe 2 seconds, there is quite a lot of FPS lag e.g. going down to 5 FPS before it then raises, as if something major is happening at the start of each motion event.

Here is my config now: https://image.ibb.co/g5QqMR/wFIyx2r1.png

Also, I've noticed that my 'Motion Gap' setting is seemingly being ignored, when there's a motion event for maybe 5 seconds, a single 15 second video gets created as per the 'Maximum Movie Length' setting but a 2nd should also be created to cover the additional 5 seconds I have set in the 'Motion Gap' setting (20 seconds).

None of these issues were present when I used your build, @jasaw however what made me switch to the mainline motioneyeos was that my raspberrypi randomly died one day and on boot it said it couldn't get any wifi network, I couldn't see any issues with my wpa_supplicant file and it hadn't been touched so I gave up and just re-installed with the mainline.

jasaw, 2017-10-29:
That's now how motion gap works. When there's a motion event, this is what happens :
1. Record captured before. 
2. keep recording until number of motion frames in buffer falls below Minimum Motion Frames.
3. Record captured after.
4. Stop recording and start counting down Motion Gap frames. 
5. If there's no more motion after Motion Gap, trailer gets written to video file. If there's more motion within the Motion Gap period, repeat step 1.

As for the video created, it depends on what your frame rate is. 

Another thing is, you might want to use the hardware accelerated encoder available on version 20171008. Choose H264/OMX as your movie format. You should be able to run 25 or 30 fps. I find Minimum Motion Frames set to 75% of the frame rate is a good start, then you tweak it and the threshold.

cuddylier, 2017-10-29:
@jasaw Thank you for the explanation, the overall process makes much more sense now.

Only thing is, my 'captured after' is 600 frames so should that not be 24 seconds or so at a frame rate of 25 meaning my recordings should last an absolute minimum of 24 seconds?

cuddylier, 2017-10-29:
@jasaw One last thing, do you know the MB RAM usage per 'captured before' frame so I can work out what my Raspberry pi 3 can cope with?

Thanks

jasaw, 2017-10-29:
@cuddylier The "captured after" of 600 frames does not guarantee 24 seconds because it depends on the actual frame rate you're getting, but I would expect the video to last at least 24 seconds.

To find out how much RAM you can spare for "captured before", the easiest way is to run "top" command, and see how much RAM motion process is consuming.
The actual amount of RAM needed depends on the resolution, image format, and the number of cameras. I'm pretty sure the raw image is stored in YUV420 format, that means 6 bytes per 4 pixels (assuming no padding and no rounding up). So for a single camera, 1920 x 1080 resolution, you end up with 1920 * 1080 * 1.5 bytes per image, which is around 3.2MB each image. "Captured before" of 200 gives you roughly 640MB. Remember to reserve some memory for other processes.

cuddylier, 2017-10-29:
@jasaw I'm only getting 15 second maximum movies in some cases for 25 - 30 FPS, any ideas why it's not creating another movie and continuing recording?

jasaw, 2017-10-30:
I'm not sure what the exact behaviour is when it hits the "Maximum Movie Length" during the motion gap period.

cuddylier, 2017-10-30:
@jasaw Ah okay, thanks for your help on my other enquiries. Maybe @ccrisan can chime in on this.

ccrisan, 2017-10-30:
@cuddylier I don't have too much experience with those timing settings in motion. At least not more than @jasaw has. I'd recommend heading over to [Motion](https://github.com/Motion-Project/motion) and ask there about these settings (which are basically transparent for motionEye).

cuddylier, 2017-10-30:
@ccrisan Thanks.

One last thing, I sometimes have movies that overrun the maximum length 15 seconds I've set, is this a bug if it's a motioneyeos thing and not motion related? The creation of new movies after it overran the maximum limited stated used to be rock solid but stopped being so when I updated recently.

ccrisan, 2017-10-30:
There's a `max_movie_time` motion config option which corresponds to motionEye's Maximum Movie Length. There's nothing that motionEye does to that value other than passing it to motion.

If you think it doesn't do its job as expected, see [Reporting A Motion Issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

cuddylier, 2017-10-30:
@ccrisan Ah okay, thanks for your help.

jasaw, 2017-10-30:
@cuddylier You mention `max_movie_time` worked perfectly when you were on my forked release. Do you remember which version was that?

Can you also tell me more about your set up and configuration. I'll try to reproduce this problem.
Which model of Raspberry Pi? Pi camera or USB camera or RTSP/IP camera? Resolution, frame rate?
Motion threshold, motion gap, captured before, captured after, minimum motion frames (if they are different from your previous screenshot).

cuddylier, 2017-10-30:
@jasaw I think I realise why it's going over 15 seconds now, it's adding the 90 frames I've for set for captured before making 17 - 18 seconds total on average. The capture after though did work on your fork prefectly when I had an issue open on it, it was with the older motion version of course.

Maybe you'd like to test the captured after?

Pi3
Pi camera
800 X 600
30 FPS (same result with 25 FPS)
2%
20
90
600
18

Thanks!

jasaw, 2017-10-31:
@cuddylier Reading through the source code, I think this is the behaviour with `max_movie_time` of 15 seconds.
1. Motion is detected, record the current time as the start of the motion event. Also write all "captured before" frames into movie file.
2. Keep recording while motion is sustained, or until "captured after" is finished, or until `max_movie_time` is reached. `max_movie_time` is reached when duration since start of motion event is 15 seconds. Note that "captured before" frames are excluded from the 15 seconds.
3. When `max_movie_time` is reached, the motion event is assumed to be finished. If there's more motion at this time, it will be treated as a new motion event, so new movie file will be created with whatever frames left in the "captured before".

This "pre-capture frames excluded from max_movie_time" behaviour is probably a bug, but I'm not sure. It's a question for motion developers.

cuddylier, 2017-10-31:
@jasaw Ah, that explains why it doesn't then create a new movie file if there was no motion for a while. E.g. if there was motion for 2 seconds, it usually just makes a single 15 (17) second movie. I'm okay with that I guess, I assume all the capture after is doing then is extending it to the full 15 (17) seconds. I only wanted it to keep recording with no motion for a while due to me often missing starts of new motion but the captured before seems to resolve that worry, would still be nice if captured after actually persisted and created a new movie file as needed to fulfil the full captured after time period but I guess a question for motion devs.

Thanks for looking into this for me.

jasaw, 2018-01-14:
Issue has been tracked down to how motion software handles video recording with max duration.

closed 2018-01-14 by jasaw