motioneye-project/motioneyeos #1210: PyPi package download 403 : SSL is required

jimmyzero3, 2017-10-28:
Hi,
I'm trying to build motioneyeos from source for raspi3 today and I encountered some problem.

When downloading python-pycurl 7.19.5.1 from pypi.python.org, it returned ERROR 403: SSL is required.
The build script then tried to get python-pycurl 7.19.5.1 from sources.buildroot.net but got an 404 not found and build fails.

It seems that PyPI dropped Http support on Oct 26, so you may have to make some changes to your build script.
Thanks!

jimmyzero3, 2017-10-28:
The error messege is as below.

> --2017-10-27 19:26:51--  http://pypi.python.org/packages/source/p/pycurl/pycurl-7.19.5.1.tar.gz
Resolving pypi.python.org (pypi.python.org)... 151.101.76.223, 2a04:4e42:12::223
Connecting to pypi.python.org (pypi.python.org)|151.101.76.223|:80... connected.
HTTP request sent, awaiting response... 403 SSL is required
2017-10-27 19:26:51 ERROR 403: SSL is required.
 
> --2017-10-27 19:26:51--  http://sources.buildroot.net/pycurl-7.19.5.1.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 176.9.16.109
Connecting to sources.buildroot.net (sources.buildroot.net)|176.9.16.109|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-10-27 19:26:52 ERROR 404: Not Found.
 
> package/pkg-generic.mk:128: recipe for target '/home/WhatAreYouLookingAt/Downloads/motioneyeos-20171008/output/raspberrypi3/build/python-pycurl-7.19.5.1/.stamp_downloaded' failed
make: *** [/home/WhatAreYouLookingAt/Downloads/motioneyeos-20171008/output/raspberrypi3/build/python-pycurl-7.19.5.1/.stamp_downloaded] Error 1

ccrisan, 2017-12-17:
Fixed via https://github.com/ccrisan/thingos/commit/c508ee6b3f8dc4a822556a66412b770b7167abfd. Changes will be merged into motionEyeOS really soon.

closed 2017-12-17 by ccrisan