motioneye-project/motioneyeos #1211: i2c-rtc-gpio not available?

cuteboi, 2017-10-28:
So, I've been working on this camera project again, and I am glad that the i2c-rtc was re-implemented for DS3231, but now I need pin 6 on the raspberry pi for boot up of the device when it has been told to `poweroff`, and to do that I would have to reassign the i2c pins to another set of pins.  Currently the `i2c-rtc-gpio` dtoverlay is not available, because dmesg will not show this message when I've assigned the pins:

```
pi@raspberrypi:~ $ dmesg | grep i2c
[    2.884187] i2c /dev entries driver
[    3.635818] i2c-gpio i2c-gpio-rtc@0: using pins 23 (SDA) and 24 (SCL)
```

This output has been obtained from a raspbian install.

the options in config.txt to enable this would be in:

```
dtoverlay=i2c-rtc-gpio,ds3231
```

To test with motioneye, you don't need an RTC connected to the pins, just that it's defined it will attempt to configure i2c and set it to the default i2c-rtc-gpio options of GPIO23 and GPIO24 ([full documentation in the overlays README](https://github.com/raspberrypi/firmware/tree/master/boot/overlays))

Could this be added in a future version?  I'd love to have the power on and power off buttons enabled. #

ccrisan, 2017-10-29:
The problem here is that, given the small partition layout that motionEyeOS uses, there's currently not enough room for all DTBs on the boot partition. Thus the OS comes with the minimum required boot files.

On the other hand, it should be possible to manually add your own DTB files by copying them over to the boot partition and use them. However, the OS firmware upgrade would overwrite them. There are plans to fix this overwriting issue some time in the near future. I'm keeping this issue open until I manage to fix the boot overwriting partition issue.

cuteboi, 2017-10-30:
Yeah, I had tried copying the overlays directory with the i2c dtbo files.  Still no go, not sure how to rebuild the system to handle these things.  Any insight on what I could be missing?  I'll try again during the week.  I decided to NOT put too many hours researching this weekend on a personal project.

cuteboi, 2017-11-02:
I'm in the process of running the buildroot for the rpi3 with vcdbg enabled, I'll try and see why it wont load the dtoverlay for the i2c, I'm not sure what I'm doing at this point, but I'll explore what venues I have to add in the overlays and move on with the project.  This build takes quite some time.

ccrisan, 2017-11-02:
@cuteboi the pycurl problem has been reported in #1210. I'll probably have to update the buildroot package. Until then, I recommend manually downloading `pycurl-7.19.5.1.tar.gz` and placing it in the `.download` folder.

cuteboi, 2017-11-02:
It seems other dtbs are not loading, because not found.  I think there may be some sort of incorrect fs layout.  Either way, I'll keep at it.   I had only added vcdbg and htop to the image, and enabled `dtdebug=1` and the dts I needed.

I'll report again once I figure out what's up.

```
001581.980: *** Restart logging
001582.007: brfs: File read: 120 bytes
001582.961: brfs: File read: /mfs/sd/cmdline.txt
001583.027: Read command line from file 'cmdline.txt'
dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
001583.305: brfs: File read: 144 bytes
001857.132: brfs: File read: /mfs/sd/kernel.img
001857.156: Loading 'kernel.img' to 0x8000 size 0x462570
001860.783: No kernel trailer - assuming DT-capable
001860.812: brfs: File read: 4597104 bytes
001864.683: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
001864.708: Loading 'bcm2710-rpi-3-b.dtb' to 0x46a570 size 0x44d8
001876.968: dtdebug: delete_node(/__local_fixups__)
001885.515: dtdebug: /aliases:i2c_vc=i2c0
001891.693: dtdebug: /__symbols__:i2c_vc=i2c0
001896.599: dtdebug: /__overrides__:i2c_vc=i2c0
001909.094: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
001914.132: dtdebug: /aliases:i2c=i2c1
001920.487: dtdebug: /__symbols__:i2c=i2c1
001925.493: dtdebug: /__overrides__:i2c=i2c1
001930.568: dtdebug: /aliases:i2c_arm=i2c1
001936.962: dtdebug: /__symbols__:i2c_arm=i2c1
001941.996: dtdebug: /__overrides__:i2c_arm=i2c1
001954.655: dtdebug: /__overrides__:i2c_baudrate=i2c1_baudrate
001968.091: dtdebug: /__overrides__:i2c_arm_baudrate=i2c1_baudrate
001968.124: brfs: File read: 17624 bytes
001968.261: dtdebug: Failed to open overlay file 'overlays/rpi-ft5406.dtbo'
001968.407: dtdebug: Failed to open overlay file 'overlays/rpi-ft5406-overlay.dtb'
001968.548: dtdebug: Failed to open overlay file 'overlays/rpi-ft5406.dtb'
001968.560: Failed to load overlay 'rpi-ft5406'
001968.719: dtdebug: Failed to open overlay file 'overlays/rpi-backlight.dtbo'
001968.866: dtdebug: Failed to open overlay file 'overlays/rpi-backlight-overlay.dtb'
001969.010: dtdebug: Failed to open overlay file 'overlays/rpi-backlight.dtb'
001969.022: Failed to load overlay 'rpi-backlight'
001969.053: dtparam: uart0_clkrate=48000000
001972.593: dtdebug: /__overrides__ has no uart0_clkrate property
001972.606: Unknown dtparam 'uart0_clkrate' - ignored
001973.145: brfs: File read: /mfs/sd/config.txt
001973.188: dtparam: i2c_arm=on
001976.432: dtdebug: Found override i2c_arm
001976.467: dtdebug:   override i2c_arm: string target 'status'
001982.443: dtparam: i2s=on
001985.786: dtdebug: Found override i2s
001985.822: dtdebug:   override i2s: string target 'status'
001989.874: dtparam: spi=on
001993.204: dtdebug: Found override spi
001993.238: dtdebug:   override spi: string target 'status'
001997.485: dtparam: audio=on
002000.956: dtdebug: Found override audio
002000.991: dtdebug:   override audio: string target 'status'
002009.261: brfs: File read: 120 bytes
002009.400: dtdebug: Failed to open overlay file 'overlays/i2c-rtc-gpio.dtbo'
002009.547: dtdebug: Failed to open overlay file 'overlays/i2c-rtc-gpio-overlay.dtb'
002009.689: dtdebug: Failed to open overlay file 'overlays/i2c-rtc-gpio.dtb'
002009.701: Failed to load overlay 'i2c-rtc-gpio'
002009.729: brfs: File read: /mfs/sd/overlays/i2c-rtc-gpio.dtb
002024.522: dtdebug: /aliases:serial0=uart1
002029.605: dtdebug: /aliases:serial1=uart0
002038.400: dtparam: uart1=off
002041.717: dtdebug: Found override uart1
002041.753: dtdebug:   override uart1: string target 'status'
002833.319: dtparam: arm_freq=1200000000
002836.699: dtdebug: Found override arm_freq
002836.750: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
002847.544: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
[... 43 lines cut ...]
002109.902: dtdebug: Found fragment 3 (offset 696)
002109.983: dtdebug: fragment 3 disabled
002110.063: dtdebug: Found fragment 4 (offset 920)
```

Uhm, the git version of motioneyeos, with vcdbg and htop added, and imported overlays seems to work correctly:

```
[root@meye-1defd578 overlays]# i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

The i2c bus 3 loaded correctly

I'm not sure what's so different from the version I built vs the build on the downloads page.  i2c bus 3 loads correctly on my version, but not on here.

Anything I can help with?