motioneye-project/motioneyeos #1213: Upload to Google Drive not working

felippe-ro, 2017-10-30:
Hello,

After switching on the toggle for "Upload Media Files", selecting "Google Drive" from the drop-down list, and entering the location in the format "\home\garden\" (with no quotes), I click the "Apply" button. Then I click the "Obtain Key" link, followed by "Test Service" button.

After a few seconds I get the message: "Accessing the upload service failed: Invalid query!".

I have tried this on two Pi Zero's with motionEye OS.

The latest log entries:

[...]
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 875, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 98, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 192, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 247, in _get_folder_id
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 259, in _get_folder_id_by_path
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 279, in _get_folder_id_by_name
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 328, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 396, in _request_credentials
Exception: Code was already redeemed.
2017-10-30 15:59:54: [motioneye]    ERROR: gdrive: failed to obtain credentials: Code was already redeemed.
2017-10-30 15:59:54: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/2017-10-30/15-59-44.jpg" with service gdrive: Code was already redeemed.
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 875, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 98, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 192, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 247, in _get_folder_id
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 259, in _get_folder_id_by_path
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 279, in _get_folder_id_by_name
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 328, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 396, in _request_credentials
Exception: Code was already redeemed.
2017-10-30 17:29:37: [motioneye]    ERROR: gdrive: request failed: Invalid query
2017-10-30 17:29:37: [motioneye]  WARNING: accessing gdrive failed: Invalid query
2017-10-30 17:29:59: [motioneye]    ERROR: gdrive: failed to obtain credentials: Code was already redeemed.
2017-10-30 17:29:59: [motioneye]  WARNING: accessing gdrive failed: Code was already redeemed.
2017-10-30 17:31:02: [motioneye]    ERROR: gdrive: request failed: Invalid query
2017-10-30 17:31:02: [motioneye]  WARNING: accessing gdrive failed: Invalid query
2017-10-30 17:39:25: [motioneye]    ERROR: gdrive: failed to obtain credentials: Code was already redeemed.
2017-10-30 17:39:25: [motioneye]  WARNING: accessing gdrive failed: Code was already redeemed.
2017-10-30 17:42:32: [motioneye]    ERROR: gdrive: request failed: Invalid query
2017-10-30 17:42:32: [motioneye]  WARNING: accessing gdrive failed: Invalid query

Hope you can help.

Thank you.

ccrisan, 2017-10-30:
Use forward slashes for location (e.g. `/home/garden`).

closed 2017-10-30 by ccrisan

jasaw, 2017-10-30:
I think we need to add Google Drive / Dropbox set up instructions to our wiki. Anyone got some spare time to write it up?