motioneye-project/motioneyeos #1215: detecting when some client is using the stream

dakipro, 2017-10-31:
Hi,
I would like to do some action when some of the clients are using the camera stream, either as video or as photos (jpeg url/stream is used on several devices as well). I couldn't find any info about it, but I think I saw somewhere some info mentioning "x clients connected" or something similar.

F.eks. I have a baby cam and an infra-red light in the room that per today must be turned on the whole time. I would like to turn on the light only if I am actually watching the stream from the camera, either accessing it via mobile, or home-automation system or tv or anything. I would also like to light a red LED on the camera itself in the livingroom, so that I know if someone from (hopefully only) my family is connected to the camera. I know I could use the motion detection, but sometimes there is no motion in front of the camera, I would just connect to it to see how is the baby sleeping.

Is there some api or via webhook or a local script or anything really, that would just say yes or no based on if some stream is active, or if the camera has active connection to some client?
In this scenario "stream not active anymore" switch can have a delay, but it is kinda important that "streaming now" is as close as possible to the real time (when checking if the baby is sleeping, it is ok to wait a second/two for lights to come on".
The lights could be controlled via gpio, python (or any other) script.

avanc, 2020-05-26:
I also need to know when the video is used to activate IR light at night.
If it is not available yet, a hint to code within motionEye would be great so I can start digging into it.

starbasessd, 2020-05-26:
[root@meye-4cec78b8 ~]# netstat -apt
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
    PID/Program name
tcp        0      0 0.0.0.0:ftp             0.0.0.0:*               LISTEN
     1115/proftpd: (acce
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN
     1100/sshd
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN
     1131/smbd
tcp        0      0 localhost:7999          0.0.0.0:*               LISTEN
     1222/motion
tcp        0      0 0.0.0.0:netbios-ssn     0.0.0.0:*               LISTEN
     1131/smbd
tcp        0      0 0.0.0.0:www             0.0.0.0:*               LISTEN
     1203/python
tcp        0      0 0.0.0.0:tproxy          0.0.0.0:*               LISTEN
     1222/motion
tcp        0      0 localhost:47488         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47482         localhost:www
TIME_WAIT   -
tcp        0    464 192.168.2.14:ssh        192.168.2.224:52234
ESTABLISHED 12045/sshd: admin@p
tcp        0      0 localhost:47486         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47484         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47502         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47478         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47480         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47490         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47500         localhost:www
TIME_WAIT   -
tcp        0  17730 localhost:tproxy        localhost:36508
ESTABLISHED 1222/motion
tcp        0      0 192.168.2.14:www        192.168.2.19:46988
 ESTABLISHED 1203/python
tcp        0      0 localhost:47498         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:47496         localhost:www
TIME_WAIT   -
tcp        0      0 localhost:36508         localhost:tproxy
 ESTABLISHED 1203/python
tcp        0      0 localhost:47492         localhost:www
TIME_WAIT   -
tcp6       0      0 [::]:445                [::]:*                  LISTEN
     1131/smbd
tcp6       0      0 [::]:netbios-ssn        [::]:*                  LISTEN
     1131/smbd

Also, python is installed.
[root@meye-4cec78b8 ~]# python --version
Python 2.7.15

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

avanc, 2020-05-27:
@starbasessd Interessting approach, didn't thought about that.

starbasessd, 2020-05-27:
There are better netstat filters. It was quick-n-dirty, and already
installed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

avanc, 2020-05-27:
I did some investigations on the connections and I can detect if somebody connects to motioneye via port 80. However, some connections stayed open after I closed the browser.

Thus, I had a look what happens on the client computer. Interestingly, systemd makes some connections before firefox does it:

    ['10.10.10.125:46656', '10.10.10.15:80', 'TIME_WAIT', '924', '/usr/lib/systemd/systemd']
    ['10.10.10.125:46658', '10.10.10.15:80', 'ESTABLISHED', '2309541', '/usr/lib/firefox/firefox']
    ['10.10.10.125:46654', '10.10.10.15:80', 'TIME_WAIT', '924', '/usr/lib/systemd/systemd']

This data was collected directly after browsing to motioneye (10.10.10.15) from the client (10.10.10.125) using firefox.

Also while closing the connection, systemd is involved:

    ['10.10.10.125:46658', '10.10.10.15:80', 'TIME_WAIT', '924', '/usr/lib/systemd/systemd']

It is the identical connection as before for firefox.