motioneye-project/motioneyeos #1216: Web UI crashes after just a few seconds

SaylorTwift, 2017-11-02:
I'm using MotionEyeOS v. 0.37.1 on a RasPi Zero W.
According to a tutorial, I used the Raspberry Pi version of motioneyeos, not the RasPi 2 or RasPi 3 versions. I have nothing else connected but my camera and the power supply which is a reliable 5V/2A which usually powers my Pi 3 Model B for multimedia streaming, so it should be fine for this.
The issue is this: When I boot the Pi, everything works fine as expected. I can ssh to it and open the web UI to change settings and whatnot but after anywhere from a few seconds to a minute after doing anything whatsoever in the UI, the camera preview turns off, replaced by a crossed out camera symbol, the UI is still responsive but as soon as I try to change a setting or press any button, a message pops up reading "An error occurred. Refreshing is recommended.". And when I refresh, the UI doesn't load because of an "ERR_CONNECTION_REFUSED" error in chrome. If I wait a few minutes and reload again, the UI may pop up again for a minute before crashing again.

I've disabled all port forwarding to the outside world and both the Pi and my PC are on the same network.

I'll attach all log files. I turned on Debug mode halfway through my trials.
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1439115/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/1439116/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/1439118/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1439119/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1439120/motion.log)

jasaw, 2017-11-02:
> Nov  2 16:45:20 meye-228b5063 user.notice motioneye: not responding for 120 seconds, rebooting

This event may have something to do with it. @ccrisan might be able to tell you more about this.

ccrisan, 2017-11-03:
The `motioneye: not responding for 120 seconds, rebooting` message appears when motionEye becomes unresponsive for two minutes, which in turn is caused by motion not being responsive, which in turn is usually caused by a fault camera (driver).

There's nothing I can do about this.

closed 2017-11-03 by ccrisan

SaylorTwift, 2017-11-03:
@ccrisan That's interesting. I just waited for the web UI to shut down and then visited the streaming link and the stream was still going and working fine. Doesn't that mean both camera and driver are okay?

ccrisan, 2017-11-04:
@SaylorTwift so you're using netcams? If that's the case, please try to run motion separately with the motionEye configuration (see [Report Motion Issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue)) for details on how to do that. You can then access the motion stream directly and see if the problem persists. If it does, please report a motion issue.

StuartOswald, 2018-01-16:
Unfortunately, I have the exact same issue as the OP. Any suggestions on how to resolve this.