motioneye-project/motioneyeos #1217: Unable to select storage device in menu

aramsichord, 2017-11-03:
Hello,

I am setting up motion eye os on my Raspberry Pi 2 with the original pi camera. I am trying to set a 32GB, USB 3.0 Kingston flash drive as the storage device, but it doesn't appear in the menu. I also have a wifi adapter plugged in. So far, I have tried different flash drives and have also set the max usb current to 1. None of this has worked.

Thanks for your help!
Here are the logs.
[boot.log](https://github.com/ccrisan/motioneyeos/files/1439731/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1439732/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/1439733/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1439734/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1439735/motioneye.log)

ccrisan, 2017-11-03:
@aramsichord what OS version do you run? What filesystem do you have on your flash drive? Have you tried other flash drives to see if they work?

As far as I can tell from your logs, the following indicate that your system detects the flash drive:

    [Wed Dec 31 16:00:03 1969] scsi 0:0:0:0: Direct-Access     Kingston DataTraveler 3.0 PMAP PQ: 0 ANSI: 6
    [Wed Dec 31 16:00:03 1969] random: crng init done
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] 60632064 512-byte logical blocks: (31.0 GB/28.9 GiB)
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] Write Protect is off
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] No Caching mode page found
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] Assuming drive cache: write through
    [Wed Dec 31 16:00:05 1969]  sda: sda1
    [Wed Dec 31 16:00:05 1969] sd 0:0:0:0: [sda] Attached SCSI removable disk

aramsichord, 2017-11-03:
I have tried other flash drives, they don't work either. Neither ext4 or FAT32 have worked. I am using the latest version of the software (20171008 for raspberry pi 2). I am not positive of the quality of the outlet and the power going to the USB ports since I am using both a wifi adapter and a USB drive, but everything else seems to be fine. 

I think the drive may have appeared during initial setup when I was using a different outlet, had no wifi adapter, and was using ethernet, but it wouldn't be easy to move the setup right now to check.

Thanks for the help, I appreciate your time!

ccrisan, 2017-11-04:
@aramsichord can you try a previous stable version (e.g. 20171003) and see if the problem persists? It may be a regression bug.

aramsichord, 2017-11-06:
Thanks for the help, I decided to just store the pictures on the sd card.

closed 2017-11-06 by aramsichord