motioneye-project/motioneyeos #1219: MotionEye saves images and mp4 but can't retrieve them

enterprisemike, 2017-11-04:
Hi,

I'd really appreciate some guidance on this one. I have MotionEye setup with an external USB drive mounted at /data/media. Image and video is being written there with no problem. I can see them via samba and SSH.
However on the MotionEye web interface, when I try to view images or video I get:
"Pictures taken by [Camera name]"
(no media files)

The File Storage settings for the cameras are:
Storage Device: WD My (/dev/sda1)
Root Directory: /footage/[CameraName]

It seems like MotionEye is looking in /data for the images.
If I change the setting to write back to what I thought was the SD card (/dev/mmcblk0p3), it writes and retrieves fine. E.g.
Storage Device: /dev/mmcblk0p3
Root Directory: [Camera Name]

Writes and retrieves from the SD Card /data/[Camera name] directory.

Can someone point me in the right direction to get MotionEye accessing the USB storage please?

Thanks
Mike

CCRISAN REPLY:

> First off, given the /data/media path, I strongly believe you're using motionEyeOS and this issue would belong to motionEyeOS instead of here.
> 
> In any case, motionEye is indeed looking in the root directory and searches for files recursively. If your media files are in /data/media/sda1/footage/[Camera Name] (which is what I assume given the details you offered here), then your Storage Device should be /dev/sda1 and your Root Directory should be /footage/[Camera Name]. So please stick with the first variant of your settings and check if the files are indeed on that path.

ccrisan, 2017-11-04:
Please enable debugging, try opening the media browser again and attach your motioneye log file.

enterprisemike, 2017-12-08:
[motioneye.zip](https://github.com/ccrisan/motioneyeos/files/1544123/motioneye.zip)
Hi, thank you. Sorry for the delay - I only get to this site once every month. The debug log is attached. 

Kind regards
Mike

ccrisan, 2017-12-11:
You have many `Input/output error` entries in your motioneye log file. This usually indicates a hardware disk problem. Here are some troubleshooting ideas:
 * make sure your power supply is powerful enough, stable and power cables are short
 * try replacing your external USB drive to make sure it's not faulty
 * remove unnecessary USB devices to reduce power consumption and fluctuations

closed 2017-12-11 by ccrisan