motioneye-project/motioneyeos #1220: Problem when adding new camera

martyouel, 2017-11-04:
Hi,

i'm using motioneye since now maybe 1 year and half and I have nothing to complaint about.
I'm using it with an older trendnet ip cam and accessing it from this URL : http://192.168.12.XXX:8085/cgi/mjpg/mjpg.cgi

I received yesterday the new camera I bought, a cheap XIAOMI. I did the "fanghack" to be able to rtsp the camera, so i'm easy able to now access the camera with this address : rtsp://192.168.12.XXX:554/unicast.

I then tried to add the camera in MotionEye.

I did : add camera / Network Camera / Entered the URL (no username, no password) and motion eye detect the camera type as "RTSP/TCP Camera".

When the camera has beed added, I now see "2 box" for the preview of my 2 cameras but I get 2 problems -->

1) The feed of my old camera is freaking up. I see an image, then it cut (no image), image back, image gone...it flicking

2) I see no feed on my new cam added

I tried the camera in a windows motion application, and the RTSP works great in this application.

Any help would be appreciated.

thanks

ccrisan, 2017-11-06:
If the RTSP stream works with other pieces of software but does not work with motion, please [report a motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

closed 2017-11-06 by ccrisan

ktpx, 2018-09-06:
I have an issue with this too. I can add the camera, and it seems OK.  It works in VLC (H264 codec), but nothing shows in MotioneyeOS. I get a grey image with "connection to camera lost".