motioneye-project/motioneyeos #1221: No Chance to get any Image from MotionEye to run for the Pi 3

OldHolborn01, 2017-11-04:
Hello,

First, for the Pi 1 B+ and P2 the Images workes very well for me!

I try out the same for Pi 3, but there are several of fails and it hangs on the login, but there are no chance because of User  and Password.

What goes wrong? (see the Logs in the Pictures)

The Hardware does function, I have tried 8GB and 16GB SD-Cards from Toshiba also with the original Image from Raspian, which is ok.
[image]
[image]

Pleasy are there any Tipps?

Thank you
Reinhard

jasaw, 2017-11-05:
Looks like a corrupted SD card. Have you tried reflashing it?

OldHolborn01, 2017-11-07:
I´ve tried a 8GB SD-Card, a 16GB and also several other Cards, which I had used for the Pi 2 or Pi1 B+

The last one, (16GB Toshiba class 10), I putted the Noobs Image on them, because of find out a hardware problem. The Installation goes without any errors, therefore I think, the Card and the Pi are ok...

I will try it again and will post my results soon...

Thank you for help.

Reinhard

OldHolborn01, 2017-11-09:
Yes, it was indeed the SD, 2 new cards are not working, but my cards that I used for Pi1 and Pi2 are running. The new cards, however, work with the normal operating systems for the Raspberry. It's very strange.

This clarifies the error and the topic can be closed.

Thank you very much for the help!

Reinhard

jasaw, 2017-11-09:
Very glad that it's working for you now. Thanks for reporting back.

closed 2017-11-09 by jasaw