motioneye-project/motioneyeos #1223: Wifi Wowes

pkwentus, 2017-11-06:
I'm trying to configure WIFI on both Raspberry Pi's   I started with a zero w and have moved on to a 3.  For some reason, I keep encountering a wpa_supplicant error (brcmfmac).  

The Rpi 3 works well with raspian and successfully connects to my network as it is configured today.  

With Motioneyes, I suspect that perhaps my router is the issue?  I'm using an Asus TM-AC-1900.  The 5ghz channel is configured with  WPA2-Personal, AES Encryption.  On this particular router, I have several options for security and encryption.  Can you please help me understand what I'm missing here?!?!

Authentication:
Open System
WPA2-Personal
WPA-Auto-Personal
WPA2-Enterprise
WPA-Auto-Enterprise

WPA-Encryption
AES
TKIP+AES

pkwentus, 2017-11-06:
Sorry- I forgot to mention that I followed the Wiki https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration to create the wpa_supplicant.conf file for both the zero W and the R Pi 3.

pkwentus, 2017-11-08:
Issue solved!
**Can you please update the WIKI, the RPi will only work with 2.4 GHz wireless.**  I was attempting to connect to the 5GHz channel of my router.

jasaw, 2017-11-08:
Thanks for the update. I've updated the [wireless network troubleshooting wiki](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems).

closed 2017-11-08 by jasaw