motioneye-project/motioneyeos #1224: [Feature Request] Fire Detection

tonkaskippers, 2017-11-06:
Hello,
I am requesting the ability to alert the user if there is either smoke detector sounds, fire alarm sounds, or fire strobes.

jasaw, 2017-11-14:
I don't think "fire alarm sound detection" will ever get implemented in this surveillance software. For something so critical like a fire alarm, it's better to have dedicated hardware that actually detects smoke. @ccrisan What do you think?

tonkaskippers, 2017-11-14:
Just to be clear, I meant this for notification purposes, not for actually
saving fires.  Just to notify that the fire alarm is going off.

ccrisan, 2017-11-14:
This doesn't seem like something that comes out-of-the-box with a video surveillance software. I don't see this ever being part of motionEye(OS).

closed 2017-11-14 by ccrisan