motioneye-project/motioneyeos #1225: Cronjob not working

Walsonde, 2017-11-06:
I added a python skript, which shuts down the Pi after Pin 3 is connected to ground. It works when I start the skript directly in the shell. But I want it to start automatically after booting the pi. So i added the line
`@reboot /usr/bin/python /path/to/my/skript/shutdown_button.py`
to the crontab, but it isn't working. Can anybody please help?

dhruva-pr, 2017-11-09:
if it is to be run once after boot, refer this [https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands)

Walsonde, 2017-11-10:
I'm a little bit embarrassed now. ._.

Thanks for your help. <3

closed 2017-11-10 by Walsonde