motioneye-project/motioneyeos #1228: Request: Network Camera with H264 stream

dhruva-pr, 2017-11-09:
Trying to connect a network Camera with input H264 stream, all works well, however the CPU usage is high. Anything beyond two cameras is streching the CPU. Is there a way to use Hardware decoder for the input stream?. The MJPEG stream is far easier on the CPU but not all IP cameras support MJPEG streams.

Thanks

jasaw, 2017-11-13:
I don't have an IP camera with H264 stream, so can't test it myself, but I think can be implemented quite easily on a Raspberry Pi.

We already have ffmpeg compiled with h264_mmal. To use it to decode H264, we could try adding something like `    rtsp_data->codec_context->hwaccel = ff_find_hwaccel(rtsp_data->codec_context->codec->id, rtsp_data->codec_context->pix_fmt);` to `netcam_rtsp.c: netcam_rtsp_open_codec` function, probably after the `avcodec_parameters_to_context` call.

Another way as a quick hack is to move the `h264_mmal` up the codecs list in ffmpeg `allcodecs.c: register_all` function, i.e. move `REGISTER_DECODER(H264_MMAL,         h264_mmal);` above `REGISTER_DECODER(H264,              h264);`

Again, I haven't tested any of these methods, so they may not work at all.

jasaw, 2017-11-14:
@dhruva-pr I did the quick hack method I suggested earlier, but don't have IP camera to test. Are you able to install my [forked experimental version](https://github.com/jasaw/motioneyeos/releases/tag/20171008-dev) to test?

dhruva-pr, 2017-11-14:
thanks @jasaw , it works and shows a significant decrease in CPU as long as the input resolution is the same as output resolution. However when using multiple IP cameras with H264 streams there seems to be a stability issue. Unsure if it is availability of GPU. Camera's are constantly getting disconnected & recovers only after system reboot. Will run it for a day and post findings.

jasaw, 2017-11-14:
Can you increase the GPU memory? I don't know what is the correct number.

jasaw, 2017-11-19:
@dhruva-pr Any update on your progress on hardware accelerated H264 decoding? I'm curious to find out if hwaccel H264 decoding works. :-)

dhruva-pr, 2017-11-22:
Hardware decoding does work. Shows significant reduction in cpu usage. yet to test on reliability with 4 h264 streams(need to test with higher GPU memory). should try over the weekend. will post findings.

gururise, 2018-05-19:
Any updates on hwaccel H264 decoding?  I'm trying to determine whether its better to use a Odroid-XU4 (presumably w/o hwaccel) or a Raspberry Pi 2 (with hwaccel) for handling 4-5 external cams.

boardtim01, 2018-10-16:
How do I implement this? What file or files need to me modified? 
To use it to decode H264, we could try adding something like rtsp_data->codec_context->hwaccel = ff_find_hwaccel(rtsp_data->codec_context->codec->id, rtsp_data->codec_context->pix_fmt); to netcam_rtsp.c: netcam_rtsp_open_codec function, probably after the avcodec_parameters_to_context call.

Another way as a quick hack is to move the h264_mmal up the codecs list in ffmpeg allcodecs.c: register_all function, i.e. move REGISTER_DECODER(H264_MMAL, h264_mmal); above REGISTER_DECODER(H264, h264);

jasaw, 2018-10-18:
@xu4user Keep in mind that this is a hack that I have not tested. You are free to test it if you want.
Here's what you need to do:
1. Git clone motioneyeos repository if you haven't already done do. You'll need to compile your own motioneyeos image for this.
2. Download this patch file: https://github.com/jasaw/motioneyeos/blob/mods/package/ffmpeg/0002-prefer-h264-mmal-decoder.patch and place it `motioneyeos/package/ffmpeg` directory.
3. Build motioneyeos image. See wiki pages on how to build.
4. Flash the image onto your SD card and test.

shahzadgodil2, 2019-01-13:
> I don't have an IP camera with H264 stream, so can't test it myself, but I think can be implemented quite easily on a Raspberry Pi.
>
> We already have ffmpeg compiled with h264_mmal. To use it to [...]

Where exactly this file exist?
allcodecs.c:

jasaw, 2019-01-15:
@shahzadgodil2 Are you building motionEyeOS from source? If you are, then you can try the prefer h264_mmal hack by downloading this patch file https://github.com/jasaw/motioneyeos/blob/mods/package/ffmpeg/0002-prefer-h264-mmal-decoder.patch and placing it in `motioneyeos/package/ffmpeg` directory, then you build the motionEyeOS image.
If you are running Raspbian, you'll need to uninstall the default ffmpeg package, download ffmpeg source, apply the prefer-h264-mmal patch, compile, and install.

shahzadgodil2, 2019-01-16:
Will creating MotionEyeOs will work on Windows? I have Windows computer

> @shahzadgodil2 Are you building motionEyeOS from source? If you are, then you can try the prefer h264_mmal hack by downloading this patch file [...]

shahzadgodil2, 2019-01-16:
I am using  MotionEyeOS image. And not from the source
https://github.com/ccrisan/motioneyeos/releases/download/20181209/motioneyeos-raspberrypi3-20181209.img.xz

shahzadgodil2, 2019-01-17:
I have created an image file for so
My SD card is 16 GB

Now I just found that your other repository have hardware decoding implementing
https://github.com/jasaw/motioneyeos/releases

My data is not important and I can reflash it
Will installing this build will resolve hardware decoding?

shahzadgodil2, 2019-01-17:
Hello,

I am getting 15 fps in Raspberry PI 3 Model B+ in this new build

Resolution: `1920x1080
Frame:  25 fps

GPU states:

throttled=0x0, malloc=8M, reloc=206M
throttled=0x0, malloc=8M, reloc=206M

jasaw, 2019-01-17:
@shahzadgodil2 How many frames per second did you get previously? Did you notice any difference in CPU load?
What's your configuration? Are you encoding the video on the same RPi as well?

shahzadgodil2, 2019-01-17:
I was getting 8 to 12 frames in an old version

Configuation:

Resolution: `1920x1080
Frame: 25 fps
Video quality: 100%
Camera just one: Xiomi Camera: Da Fang Camera 
With Dafang Hack

I am getting 25 fps in 1920x1080 in other software also

What exactly RPi is?

Thanks

jasaw, 2019-01-17:
@shahzadgodil2 Sorry, I don't quite follow. Which other software are you getting the 25 fps?

RPi is short for Raspberry Pi.

shahzadgodil2, 2019-01-18:
I am getting 25 fps in Shinobi software in my Rasberry PI 3 B+

I am getting 15 fps in MotionEye, your new software in same Rasberry PI 3 B+

jasaw, 2019-01-21:
The below is copied directly from Shinobi guide:
- No transcoding. It's simply too CPU intensive even with hardware-accelerated h264_omx encoder. For any recording/streaming you'll need to set the video codec to "copy" (or possibly the jpeg API).
- For any decoding you'll need to use the hardware-accelerated h264_mmal codec. Without specifying this codec there will be too much CPU usage. Using MMAL ensures that the heavy lifting of deciding the h264 stream is done on the GPU.

Let's take a look at the video processing pipeline in Shinobi:
```
H264 encoded stream (CPU) +--> h264_mmal decoding (GPU) --> possibly motion detection (CPU)
                          +--> storage (CPU)
```
The GPU is only tasked with decoding the H264 video stream from an IP camera, so you get higher throughput. Disadvantage is, you can't add text or timestamp into the recorded video. This is exactly the same behaviour as motion's passthrough-recording feature (motion is the underlying software in motionEyeOS).

---
This is what we have in motionEyeOS:
```
H264 encoded stream (CPU) --> h264_mmal decoding (GPU) --> motion detection (CPU) --> draw text in video stream (CPU) --> h264_omx encoding (GPU) --> storage (CPU)
```
The GPU has to decode then encode the video stream. That's double the amount of raw video going between GPU and CPU, so we are hitting the memory bandwidth, thus lower throughput. Advantage of this setup is the video can be edited in anyway we like, e.g. add timestamp, or draw red box around detected motion area.

---

Both methods have their pros and cons. Which one to choose is up to your requirements.

jasaw, 2019-10-30:
Motion now supports selecting a decoder for netcam. https://github.com/ccrisan/motioneye/issues/1519