motioneye-project/motioneyeos #1229: macOS - any way this can be installed on a Mac?

jat80, 2017-11-09:
Just wondering if I can install this software ON a Mac?

joshlukas, 2017-11-17:
Hi. Motioneyeos is a complete operating system based on a Linux kernel. Maybe there is a way to install motion and motioneye as a frontend on macOS, but I don't think so.

tomwoodhams, 2017-11-17:
You could probably do it in parallels or similar, but would be a bit inefficient.

jasaw, 2017-11-19:
@jat80 You're better off install motionEye on your Mac, rather than motionEyeOS. This [motionEye installation wiki](https://github.com/ccrisan/motioneye/wiki/Installation) might be helpful.

jat80, 2017-11-19:
@jasaw I’ve looked throughout the wiki, issues and generally on google search and can find nothing on how to install motioneye on a Mac.  If you know of how please let me know but I have come to the conclusion it cannot be done

jasaw, 2019-10-30:
Closing issue due to lack of interst.

closed 2019-10-30 by jasaw