motioneye-project/motioneyeos #1230: MotionOs does not STOP streaming, what to do?

elrinz, 2017-11-09:
First of all, I really like Motion Eye OS. Its a great tool and well made.
I have a RPi 3, with a simple iBuffalo camera. This works well. 

My main problem is that I cannot STOP STREAMING. So, I would like to stop streaming, so that nobody who types in the address of the device will see the camera. I just want to record motion activated video. 

I tried the last few versions 10.08 down to the June version, but for all of them if i turn streaming OFF it just keeps streaming like nothing happened. Additionally, I tried to put password authentication (direct or the other one) but doesnt do anything. I do not need to be authenticated to see the webcam stream. So, what is going on? what am I doing wrong.

Thanks a lot for your reply.

Miep

elrinz, 2017-11-13:
Hello again,

So, I again tried many different versions, but all of them when I turn off streaming, actually do not STOP streaming (yes, I did use apply, and restarted the pi, etc). So, if I type in the url (e.g. 192.168.1.100 or something) then the camera still appears like nothing happened (so, when streaming is OFF). Also, it does not matter if I type in an "authentication" option, I actually do not need to authenticate to see it. I re-installed this with the last 5-6 versions (10.08 and below), all of them have the same thing (at least here). Probably its my set-up.. but no idea what I am doing wrong, or why this is happening. 

[image]
[image]

Could you please tell me why this is happening. I really have no clue what I am doing wrong. I simply want to record video when movement is detected, and have no online streaming. 

Thank you for your answer in advance.

Kind regards,

Miep

shyam4290, 2017-11-13:
Hello,
      By switching off Video Streaming you only stopped the streaming on port like (192.168.1.100:8081).If you don't want others to see the camera then protect it with password.
[image]