motioneye-project/motioneyeos #1233: No wifi with motioneyeos-raspberrypi3-20171008.img

Skydive1, 2017-11-10:
Hello everyone,
 that's my problem: I installed the version 20171008 on my Rpi3. When I plug in my power, the process takes place and when we get to the wire connection line, it puts me failed.
I looked at the SD card the file after installation via win32diskImager: I've 
bcm2710-rpi-3-b.dtb
bootcode.bin
cmdline.txt
config.txt
fixup.dat
fwupdater.gz
kernel.img
start.elf
I noticed then that I did not have the file concerning wpa_supplicant nor file named network ... should I add them? How to remedy this. 
Thank you

tuppes, 2017-11-10:
Just follow the instructions here:
[https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration]
Good luck!

Skydive1, 2017-11-12:
I did not follow these instructions, but I still managed to run motioneye on Wifi. First, I launched the rpi ethernet. Then I configured the wifi inside the interface. Finally, I reboot the system by unplugging the ethernet cable and that's it, it is configured in wifi. Thanks a lot

jasaw, 2017-11-12:
@Skydive1 Issue resolved?

Skydive1, 2017-11-14:
Yes i've resolved it

closed 2017-11-14 by Skydive1