motioneye-project/motioneyeos #1234: Wlanstick not works in Raspberry Zero with Motioneyeos

davimas, 2017-11-10:
Hey,

the wlan connection from ZeroW is bad, and i buy 2 different USB Wlan sticks with Micro USB Adapter to became more Speed with Raspberry Pi ZeroW.

On Raspberry Pi3 both works, plug and play..........but not on ZeroW
Must i make anything manually on zero?

I use the latest Motioneyeos Version on RPI3 and ZeroW

davimas, 2017-11-12:
today i instal Motioneye on Raspbian Jessy on the ZeroW and the Wlan USB Adapter works.
But same thing, the performance on Motion is the same. Low and stocking.
The extra USB Wlan Adapter not help.
In the Fast Camera Setting, works good, but when you switch to Motion.....terrible.

Settings 1600*1200/25fps
All under it, is not 2017 worthy ;-)

jasaw, 2017-11-12:
In regards to your Wifi USB adapter not working on Pi Zero W, I have never tried this configuration, but @ccrisan might be able to help you out.

About the performance issue, what exactly are you trying to achieve and what performance are you getting? What is your configuration? Local motion triggered recording or Fast Network Camera mode?
1600 x 1200 resolution at 25 fps is too much for a Pi Zero W to handle.

ccrisan, 2017-11-13:
@davimas does your setup work fine with Raspbian on the Pi ZeroW? Have you blacklisted the ZeroW internal network adapter so that yours is actually used?

davimas, 2017-11-13:
Hey, yes, the configuration with 1600*1200 with Motion is to much for the Zero. When i set him in Fast Camera Mode, works fine with this ScreenSize.

On ZeroW with Raspbian and Motioneye the USB Wlan Adapter works, he was listed as wlan1 under lsusb and dmesg. 

I only delete the wlan0 in the Config and the wlan1 works alone

jasaw, 2017-11-13:
Issue resolved?

jasaw, 2018-01-14:
Closing issue due to lack of update. Issue seem resolved anyway.

closed 2018-01-14 by jasaw