motioneye-project/motioneyeos #1235: Load screen showing lots of failures

hongomg, 2017-11-12:
Hi

Thanks for the great work, I tried getting this working on my raspberry pi 3, made the SD card following the instructions online and it boots but showing lots of failure messages, and also I can't seem to find it in my home network afterwards, so I suspect I've installed it incorrectly.

Please see screenshot of boot screen on the pi: https://imgur.com/a/rzrIt

OldHolborn01, 2017-11-13:
same Problem for me, ccrisan says, it´s the SD-Card, try another one, this would help.
[https://github.com/ccrisan/motioneyeos/issues/1221#issuecomment-343320586](url)

hongomg, 2017-11-13:
I put a samsung evo card in and it worked! thanks for the help

jasaw, 2017-11-13:
Looks like an SD card issue and changing SD card resolved it. Closing issue.

closed 2017-11-13 by jasaw