motioneye-project/motioneyeos #1237: I can connect through Putty, but I get ERR_CONNECTION_REFUSED in Chrome

patrickdemers6, 2017-11-13:
When I open a connection to the device through Putty, it says: 

> login as: admin
> Welcome to meye-43b1874d!
> [root@meye-43b1874d ~]#

After a few minutes, it will say "Server unexpectedly closed the connection"
What is the issue? Thanks!

cuteboi, 2017-11-16:
Do you have any cameras connected to the device? My RPi3 fails to have a web server running at times when there is no CSI camera. Remotely connecting via SSH is very handy.

jasaw, 2018-01-14:
@patrickdemers6 You didn't provide any details on your setup, so there's not much we can help. I'm guessing your pi lost its network connection and rebooted itself after a while. It's expected behaviour for motionEyeOS to reboot if it loses network connection.