motioneye-project/motioneyeos #1238: Hanging on ftp error

va24531, 2017-11-14:
I have the latest MotioneyeOS running on a raspberry pi 3 with 2 ONVIF cameras, works great, love the system. I have it set up for management at our business to monitor the cameras via intranet on site, and to upload images at 15 second intervals to our remote FTP server for offsite backup (I have the system setup to save the last 2 days of image captures locally as well). Everything has been working flawlessly for months now, or so I thought...

We have wifi internet through the county, and occasionally it will drop out for a few seconds. I have another RPi in front of the motionEyes RPi acting as a bridge, and then router coming off of that give intranet to the building. The bridge acknowledges the WLAN side drops out and reconnects to it as soon as wifi signal is back (usually a few seconds, not even long enough for streaming services to go into buffering mode) and leaves the eth0 side up constantly, as it should.

The issue we're having is that occasionally, when our internet connection drops out and MotionEyes is hitting the connection to upload the two images (about 200k each), the FTP client of MotionEyes doesn't acknowledge that the connection died (As if the connection is dropped half way through the transfer) and sits there and until I manually restart it.

We're using ProFTPD on the remote end, and I can verify (through netstat on the server side and local router logs) that once the ftp side of MotionEyes goes belly up, it does NOT try to re-establish connection and pick back up, so I'm 98% sure it's not on the ftp server side. **As a side note, if the connection drops BETWEEN image upload cycles, all is good, MotionEyes picks back up and uploads the ones that it couldn't ... so the ONLY time it chokes is when MotionEyes's ftp connection dies DURING a transfer cycle.

Of course I could do a VPN and bring up the UI for MotionEyes remotely to reset it, I could call and have staff onsite reboot the system IF I gave them the password, I could slow down the transfer cycle, I could do a lot of things, but I'm restrained by management as they pay my bills :)

It looks like the only option I have is to cron a reboot ever hour, so at the most we'd loose 59 minutes of archives instead of the 2 or 3 days that sometimes happens now (As I'm currently on-site 2 to 3 days a week, unless the SHTF.)

Anyone have any other idea's that could be used in conjunction with what's already provided by MotionEyesOS, software wise? I don't want to forcefully install anything else so that it's not a clean system anymore.  Anyone know of any config settings (system wise or UI wise) that could be tweaked? Any help is appreciated.

VA24531

jasaw, 2017-11-14:
I'm seeing FTP upload issues as well on motionEyeOS, and it's not an easy problem to solve. I'm thinking of using rsync instead of FTP.

If you must use FTP, maybe someone who has more knowledge in this area can help.

va24531, 2017-11-14:
Never one to complain about FOSS, if someone (ccrisan) could put a watchdog or catch-22 in place that if a FTP upload times out/errors out/fails to complete, it would just kill that process after 30 seconds or something, that may 'patch' the problem, since it seems like a ftp client problem and not necessarily a MotionEyes 'problem' per say.

Chiny91, 2017-11-14:
I don't use ftp (vastly preferring rsync) so can't check but ISTR that motionEyeOS has an ftp server.   If the @va24531 system has an ftp client, then maybe a motionEyeOS cronjob to ftp the 2 * 200k pics would do.   The crontab would include the busybox timeout command, set to just exceed the pics upload time.
`mins hours * * * timeout -t 30 /usr/bin/ftp [options and so on]` 
If the problem is the motionEyeOS client as you suspect, the problem might vanish without `timeout`.

I also trap any dud files (typically of 0k, or over the maximum likely size), throttle the bandwidth (where R Pi are a bit limited) and nice the process (new pics need cpu more than backing up pics). 

I recently abandoned 2 * R Pi 3s for 2 months, each handling 3 IPcams and rsync backup to a NAS.   Not a murmur of a backup problem, either then or in the subsequent weeks.   If mgt won't pay, rsync won't help but at least you will be thought pro-active (my corp-speak may be dated).

va24531, 2017-11-14:
@Chiny91 - Thanks for your insight. I guess a route I can go would to be spin up a VM box for the MotionEyes to FTP into on the local net, so that I can bypass the connection issue, then have the VM Box rsync the contents into the remote server, since the VM box will be free to install software on. Then, after more testing, move the VM setup over to a Pi and let it rip from there. I can write off a $50 pi and sd card and keep on going.

I've just been pulling hair out trying to figure out a way around or through this issue.

Although I wouldn't mark this issue 'fixed' - I would certainly say that having MotionEyes dumping to local and then sending from a second machine is certainly one way to skin the cat :)

Thanks for you input!