motioneye-project/motioneyeos #1241: Unable to update firmware

rjpontefract, 2017-11-15:
I tried updating MotionEyeOS running on a RPi 3 from 20161125 (4.0.1+git37b3595) as reported by the UI.  Hitting "Software Update: Check" says that version 20171003 is available, so I clicked on "Yes" to update.  It took a few minutes, rebooted and told me that the update was successful, however the UI still reports the original version.  Here's what the log contains, I don't see any errors:

> 2017-11-15 23:19:06: [motioneye]     INFO: updating to version 20171003...
> 2017-11-15 23:19:06: [motioneye]     INFO: killing motioneye init script...
> 2017-11-15 23:19:06: [motioneye]     INFO: stopping netwatch init script...
> 2017-11-15 23:19:06: [motioneye]     INFO: downloading https://github.com/ccrisan/motioneyeos/releases/download/20171003/motioneyeos-raspberrypi3-20171003.img.gz...
> 2017-11-15 23:19:36: [motioneye]     INFO: decompressing /data/.firmware_update/firmware.gz...
> 2017-11-15 23:20:04: [motioneye]     INFO: reading partiton table...
> 2017-11-15 23:20:07: [motioneye]     INFO: extracting boot.img...
> 2017-11-15 23:20:08: [motioneye]     INFO: extracting root.img...
> 2017-11-15 23:20:27: [motioneye]     INFO: backing up /boot/config.txt
> 2017-11-15 23:20:27: [motioneye]     INFO: unmounting boot partition...
> 2017-11-15 23:20:27: [motioneye]     INFO: installing boot image...
> 2017-11-15 23:20:46: [motioneye]     INFO: mounting boot partition read-write...
> 2017-11-15 23:20:56: [motioneye]     INFO: restoring up /boot/config.txt
> 2017-11-15 23:20:56: [motioneye]     INFO: preparing to boot in fwupdate mode...
> 2017-11-15 23:20:56: [motioneye]     INFO: rebooting...
> 2017-11-15 23:21:00: [motioneye]     INFO: interrupt signal received, shutting down...
> 2017-11-15 23:21:00: [motioneye]     INFO: server stopped
> 2017-11-15 23:21:00: [motioneye]     INFO: tasks stopped
> 2017-11-15 23:21:00: [motioneye]     INFO: smb mounts stopped
> 2017-11-15 23:21:00: [motioneye]     INFO: bye!
> 2017-11-15 23:21:27: [motioneye]     INFO: hello! this is motionEye server 20161125

Can anyone help me figuring out why it didn't update please?

jasaw, 2018-01-14:
Version 20161125 might be too old to do a firmware upgrade.

rjpontefract, 2018-01-15:
Maybe so.  It used to work, but I guess I may have left it too long since I last updated it.  I'll have to get the ladder out and do it manually.