motioneye-project/motioneyeos #1242: Excessive pictures ~ 16,000 per day looking @ my back yard.

pkwentus, 2017-11-16:
Rpi 3 with USB web cam

I'm trying to set up motion but for some reason, I get an excessive number of pictures.

I notice in STILL IMAGES section, Capture mode is set to Motion Triggered.  If I change to 
Motion Triggered (one picture), I get nothing.

I've played with settings in MOTION DETECTION... no luck.

Can you please recommend good settings or point me towards a resource?

AndyMAu, 2017-11-17:
I feel your pain. I have had similar issues in the past. I don't have a solution for you. One thing I find frustrating with motioneyeos is there are no real examples of settings to use to see what a user would like on their setup.

va24531, 2017-11-17:
Not trying to come across as rude, but you didn't really tell us how you want it to work. Do you want 1 image every 5 minutes regardless of motion, do you want 1 image when there IS motion, do you want stills/video for the duration of the motion? How you set it up depends on your needs/wishes.

16,000 images in 24 hours is about 11 images a minute, which is a LOT if there's no motion ...

So, check to see if there IS motion, or if the threshold is set too low ...
 Turn 'Motion Detection' 'ON' and under that, turn 'Show Frame Changes' to 'ON' ...
This should put a bounding box around the area that it see's motion overlayed on the real time image. Adjust the 'Frame Change Threshold'  >UP< until the 'tracking box' settles down and the image border is not RED anymore. Turning 'Auto Noise Detection' ON may help, maybe not, since your camera is not a USB camera and probably won't have much noise introduced in the ribbon cable between the Pi and the camera.

Under 'Motion Detection', values of 10/2/4/1 are a good starting point, and are the defaults for the gap/before/after/minimum frames.

(TIP: the tracking box shown by 'Show Frame Changes' will be saves to the images, so turn it OFF after you get everything set the way you want it :) )

Turn OFF Video Streaming, as it's not needed for testing and setting up. ( I don't even have it turned on at all because if I want to see the real time video, I use the main GUI window to see it, no need for the Pi to do extra work!)

Under 'Still Images', 'Capture Mode' should be set to 'Motion Triggered' and 'Preserve Pictures' is your choice.

Once you have the sensitivity set high enough ( Motion stops 'tracking' non-moving objects )  you can then fine tune it, to where it's just at the tipping point between motion and no motion. Then, back under 'Motion Detection' go down to 'Mask' as chance it to 'ON', and 'Editable' under mask type. Now you can train MotionEyes to ONLY watch certain parts of the image. Paint the areas you DO NOT want it to watch ( make trees, bushes, sky, etc. DARK, so it does NOT watch them).  After you SAVE MASK, you can then adjust the threshold down even more -- IF NEEDED -- but if you adjust it too low, you'll be back the same point again, saving bogus images.

If you need further help, feel free to ask, more than happy to help, but let us know what you've done, what the results where, and what settings had good/bad effect for you!

Hope this helps!

pkwentus, 2017-11-18:
Ok- first let me say THANK YOU!  your walk through and suggestions are a big help. 

My scope is to put this device Outside watching my driveway.  I'd like to capture an image of any vehicle - specifically the license plate as it travels up the drive.  I'm considering building perhaps a mock bird house or similar decoy to house the Pi & camera so that it can be placed in a way to achieve a good picture.  This might seem a bit weird but if you were enjoying my view right now, you'd understand :)   Lastly, I intend to send the photos to Google Drive so that if I have an issue they are preserved.

You have addressed many of my questions above and I'm testing today with your recos'.  I failed to mention in my initial post that I had the camera watching simply out the window when all of those photos were taken.  The SCREEN on the window ~might~ be causing some issue as well... (sorry for not disclosing that).  

Can you comment about automatic brightness setting for the camera?  I believe that one of my issues is that this camera is adjusting itself, which causes image change.  At the moment, I'm using this camera: Amazon Link: http://a.co/enPf1Jn

Are there any detailed instructions out there about Motion notifications?  This seems Really powerful.... 

Thanks again for your coaching and suggestions!

va24531, 2017-11-18:
Have a way I can contact you directly so not the keep this thread open? There are a few possibilities and other software available for your potential task that could be better handled in direct conversation, and I'd be more than happy to help you.

pkwentus, 2017-11-19:
Sure- I'm always open and I'd love this to work!   You can email me directly at my username @gmail.com

closed 2017-11-19 by pkwentus

russelltrafford, 2019-12-19:
I know this is from a while back, but if you can post an update as to what you two discussed offline, and what you eventually did for your situation, that would be helpful to others who are looking at a similar kind of use case. Thanks.