motioneye-project/motioneyeos #1243: Upload to Google drive has stopped from a week ago.

AndyMAu, 2017-11-17:
I noticed no more video files and no more images being uploaded to my Google drive since November 10th. Uploads to my gmail are unaffected. 
I thought it was a space issue with Google drive but plenty of Gigs available.
As a test I attempted to test service for file storage and motioneyeos timed out.
I have the log files saved ,before I attempted obtaining another key.
Log files are waiting to be sent.
Thanks again.

AndyMAu, 2017-11-19:
I'm answering my own question here. Not sure why the devs/mods at least give us a clue even if obvious to them.
I rebooted motioneyeos via software and now uploading.
I noticed several people having similar issues and none of them being addressed.
I guess case closed. :)

closed 2017-11-19 by AndyMAu

jasaw, 2017-11-19:
@AndyMAu Sorry man, I guess all the devs are busy, at least I am. We haven't responded to your issue because we haven't got time to investigate the problem. It's still good to find out why it stopped uploading in the first place.