motioneye-project/motioneyeos #1245: snupshot URL wird in motioneyeos nicht angezeigt

otti123, 2017-11-18:
Habe motioneyeos in der Version 

motionEye Version | 0.37.1
-- | --
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008
auf meinem Raspberry Pi 3 installiert. Update zeigt neuesten Stand an.
 Unter der Rubrik Video Streaming müsste eigentlich die URL für den Snupshot URL angezeigt werden. Auch ist die Funktion zur Installation einer neuen Kamera nicht vorhanden?

Woran liegt dies?

Otti123

jasaw, 2017-11-19:
Please write in English.

otti123, 2017-11-19:
Hallo and sorry, my English is verry bad. My question: i have installed motioneyeos in the version 
0.37.1 Bewegungsversion | 4.0.1 + git2a140fa
OS Version | BewegungsEyeOS 20171008 on my Raspberrypi 3
Under the column video Streaming would have to be indicated, actually, the URL for the Snapshot URL.
However, it is not indicated there.Follower is indicated under this column:
Follower is indicated under this column: 
Video Streaming 
Streaming URL 
Embed URL 
How do I find the Snapshot URL? 
Nevertheless, it would have to be indicated at this point? What can be the cause for it and what I must do, so that the Snapshot URL are indicated there.
Greeting 
otti123

otti123, 2017-11-19:
Sorry, it must be openedd

Chiny91, 2017-11-19:
Let us check the basic stuff.   You have "switched on" the menu item "Video streaming" ?    It should look something like this:
[image]
This menu item also has to be "opened" so the little triangle (top right corner) points downward.   This should show the snapshot URL (clickable link).    Do you have all this ?

otti123, 2017-11-19:
Hello, and thanks for the answer. Thus the menu point looks to me:
[image]
I use a Raspberry Pi3 with a connected PI 2 Camera.
I have downloaded this file from github. com for raspberry Pi3 and have installed on my Raspberry. 

otti123

otti123, 2017-11-19:
My video device looks thus:
[image]
[image]

otti123, 2018-01-10:
Hallo
habe pausiert und möchte jetzt weitermachen. In MotionEYE Os wird mir jetzt die snapshot URL und die Streaming URL sowie die Embed URL angezeigt.
[image]

 Ich möchte die Pi2 Kamera jetzt in das Programm Instarvision integrieren. Hier habe ich bereits zwei Kameras im Dauerbetrieb laufen.

Im Einrichtungstool für eine neue Kamera bei Instarvision habe ich folgendes Eingabebild:
[image]
Die Snapshot URL in MotionEyeOs sieht wie folgt aus:
http://192.xxx.x.xxx/picture/1/current/?_username=admin&_signature=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Wenn ich die Daten eingebe, wird keine Verbindung aufgenommen.
Gibt es da noch einen anderen Pfad oder wie müsste ich die Angaben in der Kameramaske tätigen?

Mein English ist sehr schlecht, deshalb hier in Deutsch.