motioneye-project/motioneyeos #1246: BLUETOOTH

huvarda, 2017-11-19:
HOW CAN I USE BLUETOOTH WITH MOTIONEYEOS

Chiny91, 2017-11-19:
I see this is your second request but really you will have to give a little more.   What are you trying to achieve ?   What hardware are you using ?

I haven't looked through the motionEyeOS code to check but commonly you have to set a flag somewhere to enable bluetooth.   Not all boards supported by motionEyeOS have bluetooth built in.   So, my R Pi 3s don't have anything in /etc/bluetooth; have you looked there ?

jasaw, 2017-11-19:
This is a duplicate of #1231, and possibly duplicate of #1083, all created by @huvarda. Please do not spam. If you don't get a response for your issue, it means all motionEyeOS developers are busy. I'm not familiar with Bluetooth on motionEyeOS, so I can't answer your question. Hopefully someone else can answer your question.

closed 2017-11-19 by jasaw

huvarda, 2017-11-20:
Sorry for spam, I just really wanted an answer.