motioneye-project/motioneyeos #1247: Request: Scheduled reboot

The-Excel, 2017-11-20:
I have a Raspberry Pi 3 and two USB cameras and a USB flash drive attached on a hub. Every few days, the cameras will stop responding but the OS isn't affected, and there doesn't seem to be any options to send notifications for when this happens. I'm using a 2A adapter for the Raspberry Pi so there shouldn't be any issues with power, but the cameras will still stop working and even replugging them doesn't help while it's online. I find the only way to get the cameras working again is to reboot the board, so I would like the ability to reboot daily at a scheduled time.

jasaw, 2017-11-20:
Set up a cron job that reboots that whichever time you want. https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

jasaw, 2018-01-14:
Due to lack of update, I presume setting up a cron job is good enough. Closing issue.

closed 2018-01-14 by jasaw