motioneye-project/motioneyeos #1249: Difference between H.264 and H.264/OMX Option?

helmar74, 2017-11-22:
Hi, 

in the gui I can set as movie format H.264 or H.264/OMX...
Whats the difference between those options?

Thanks a lot in advance

Helmar

jasaw, 2017-11-22:
OMX version uses the GPU to accelerate the H264 video encoding. If you encode more than 1 video stream, make sure to increase the GPU memory.

tigrrr, 2017-11-22:
To avoid opening a new issue: Choosing H264 or Matroska OMX encoder, motioneye doesn't record the video files whereas regular H264/Matroska works fine. The system is rpi3 on debian stretch, all libs are updated. What did I miss ? Thanks.
FFmpeg version:
ffmpeg version 3.2.8-1~deb9u1 Copyright (c) 2000-2017 the FFmpeg developers                                                                                                                  
built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1) 20170516                                                                                                                                       
configuration: --prefix=/usr --extra-version='1~deb9u1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping
 --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-lib
flite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-
libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-li
bvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --
enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared

helmar74, 2017-11-23:
Thanks a lot

closed 2017-11-23 by helmar74