motioneye-project/motioneyeos #1250: Design Question RpPi3 or RPi Zero W?

helmar74, 2017-11-23:
Hi,

I'm currently playing arround with MotionEyeOS. I'm planning a surveillance system for my Home.
So I have several questions which aren't anserwed in the Wiki. Maybe somebody can help?

1. I want to run a central Surveillance Server, probably based on RPi3. All cameras should send there livestream there, a external HDD(or SSD) is used for recording the streams, when motion is detected.
Is a RPi3 the right Hardware for motion detection and recording of 3-4 other MotionEyeOs Cams?

2. As Webcams (as mentioned before 3-4) I also want to use MotionEyeOs Cams. I'm not sure at the moment, which Hardware I should use. I'm thinking of RPi3 or RPi Zero W as a Board. I've got good impressions of RPi3. For RPi Zero W I'm not sure, which framerates I can get (I have a good Wifi infrastructure with a separate "Webcam Wifi"). I think of streaming 1600x1200 to the central system.
Which framerates will be possible with this resolution with a ZeroW?

3. I'm also thinking of running a Pi Zero " in "Fast Camera Mode". Is it possible than to do motion detection on the central server, which receives the stream? I tried it, but there were no menus for motion detection after switching the Rpi Zero from "normal" MotionEyeOs to "fastcamera mode".

4. In some other forums I read, that the onboard Wifi Modul of RPi3 and RPi Zero W isn't much good. Is it worth using an other Wifi Dongle (USB based) to get higher framerates especially for Zero W?

Thanks a lot in advance.

Regards 

Helmar

StuartOswald, 2018-02-16:
This is exactly the type of setup I am after. Did you find answers to your questions?

ccrisan, 2018-02-19:
@helmar74 

> 1. I want to run a central Surveillance Server, probably based on RPi3. All cameras should send there livestream there, a external HDD(or SSD) is used for recording the streams, when motion is detected.
Is a RPi3 the right Hardware for motion detection and recording of 3-4 other MotionEyeOs Cams?

Probably yes, if you're fine with relatively lower resolutions and framerate (around 1024x768@10fps).

> 2. As Webcams (as mentioned before 3-4) I also want to use MotionEyeOs Cams. I'm not sure at the moment, which Hardware I should use. I'm thinking of RPi3 or RPi Zero W as a Board. I've got good [...]
Which framerates will be possible with this resolution with a ZeroW?

I would advice against streaming video over WiFi: it constantly eats up a considerable amount of bandwidth and can be jammed. However, if you have no wired solution, the Zero W (in FNC mode) could help you reach high res and fps, if you diminish the JPEG quality enough. In the end, it's a matter of bandwidth so you'll have to experiment with these 3 options: resolution, framerate and image quality.

> 3. I'm also thinking of running a Pi Zero " in "Fast Camera Mode". Is it possible than to do motion detection on the central server, which receives the stream? I tried it, but there were no menus for motion detection after switching the Rpi Zero from "normal" MotionEyeOs to "fastcamera mode".

It is possible and it's in fact the recommended setup. If you saw no "menus" on the central server, you probably added your Zero camera as a Simple MJPEG Camera, whereas you should have added it as a Network Camera.

> 4. In some other forums I read, that the onboard Wifi Modul of RPi3 and RPi Zero W isn't much good. Is it worth using an other Wifi Dongle (USB based) to get higher framerates especially for Zero W?

No idea, honestly. Again, I tend to stay away from streaming over WiFi.

helmar74, 2018-02-19:
Hi,

thanks for the infos. I'm running a "Master" currently with 5 connected cameras (4 RPi3, 1 Zero W, 1 other Network camera). 4 of the MotionEyeOS Cams are connected via WLAN. I created a separate "Webcam WLAN". 

It's working  well with this setup. So Streaming via WLAN works proper.

Regards Helmar

closed 2018-02-19 by helmar74