motioneye-project/motioneyeos #1251: How to reach the Linux client GUI from other machines + oustside my network?

BdgBill, 2017-11-26:
Ok, complete Linux NOOB here but I'm really good at copying and pasting commands into a terminal window.

I have had a few Pi Zero's running MotionEyeOs for a while now and wanted to change my system to the "central server" setup.

Everything seemed to go fine. I installed Lubuntu on an old netbook. I have the familiar MotionEye GUI running on  the netbook and all my cameras visible. What I don't understand is how to get to this GUI from other machines. On the netbook, I reach the GUI by typing 0.0.0.0:8765 into a browser. This does not work from any of the other computers on my LAN. I also tried port forwarding the same address and nothing. I have searched through the Wiki and Google but maybe using the wrong search terms so sorry if I missed anything obvious.

Chiny91, 2017-11-26:
Hmm, well 0.0.0.0 is not a valid network address, although it does have some uses in networking, probably not an issue for you here.   All your local machines (router, R Pi, netbook, other computer) will have an IP address.   If you have not specifically set these yourself, then one of your machines (likely your router) is allocating IP addresses.   You need to use this IP address (not 0.0.0.0) in your browser to reach the GUI.   It should also be port 80, not 8765 by default.

How to find your motionEye, well the motionEyeOS Wiki helps [https://github.com/ccrisan/motioneyeos/wiki/Installation(url) or this will give you more to Google, although maybe this is a motionEye question not motionEyeOS.

BdgBill, 2017-11-26:
@Chiny91 Thanks for responding. I have been through the entire MotionEye Wiki. I followed the install instructions from there and everything went great. There doesn't seem to be much on the Wiki about actually using MotionEye after install is done though. Trying to search Google for issues with MotionEye results in almost all MotionEyeOS responses.

The GUI looks like what I'm used to seeing on the Pi Zero's but no Networking or Expert Settings tabs. I have tried entering the Lubuntu netbook's IP address with port 80 and with 8765 from other machines on my network with no luck. I have tried the 0.0.0.0:8765 as well. The only way I can see the GUI is by pointing a browser to 0.0.0.0:8765 and that only works on the netbook. 

I'm sure I'm missing some bit of Linux networking common knowledge or something. This is the first major snag I have run into on this project. The Linux box may have been a bridge too far. Maybe I will order a Pi3 to serve as a hub....

Chiny91, 2017-11-27:
If you have setting tabs missing, then you need to set "Advanced settings" under "General settings".

An IP of 0.0.0.0 is unroutable; it cannot go anywhere.   You must use something else, example (only an example, not your specific case) 192.168.1.12 for a typical home internal address.   Have a look on your network for what is issuing IP addresses, the "DHCP" tab of your router perhaps, or maybe something in the Lubuntu network settings.   This is a network issue though, not motionEyeOS and you probably need a network forum.

If you look in the messages.log (under "Expert settings"), you may well find some warnings/errors which will give you a further idea about what is happening.

BdgBill, 2017-11-29:
@Chiny91 - Thanks for your help, I got it working. Needed to replace 0.0.0.0. in the motioneye.conf file to the machine's internal IP, then forward the local IP and port 8765 in my router. You were right. Not really a MotionEye (or especially a MotionEyeOS) issue.

closed 2017-11-29 by BdgBill