motioneye-project/motioneyeos #1252: Config rtsp:// rohs ip cam.. unable to open device

daniele777, 2017-11-27:
Dear friend i really fo crazy to find the right adress for streaming...of this ip p2p cam..
i try that ..

rtsp://my_local_adreess_ofcam/cam1/h264
rtsp://my_local_adreess_ofcam/h264/ch1/main/av_stream
rtsp://my_local_adreess_ofcam/MPEG-4/ch1/main/av_stream
rtsp://my_local_adreess_ofcam//Streaming/Channels/1
rtsp://my_local_adreess_ofcam/ROH/channel/1
rtsp://my_local_adreess_ofcam/user=admin_password=mypassword_channel=1

Replace with my my_local_adreess_ofcam to ip address of cam in router... but nothing ... meyeos say unable to open device...

Need help!!! Thanks in advance!

R1NEJ0, 2017-12-04:
I'm having the same trouble... I could connect one camera but I have a pair more and it's impossible... The motionEyeOS config detect my ip camera but it says unable to open device... :(

[image]

ccrisan, 2017-12-04:
@R1NEJ0 it's very unlikely for a RTSP url to be that simple. motionEye does a simple RTSP tests and looks for some headers to verify that the URL is valid. It can't test the actual stream however. I recommend that you make sure the URL is indeed a correct RTSP url, by adding it to a video player, such as VLC.

R1NEJ0, 2017-12-04:
@ccrisan Good to your very fast answer!!! Yes, I can open the stream with VLC correctly with that URL, VLC asks me about username and password (admin - blank). I've tryed with /onvif1 on vlc and work great as well, but.. if I try tu open the stream with random words it works too... 

Thanks in advance!!

ccrisan, 2017-12-04:
If the URL works with VLC but does not with motionEye, please report an issue to motion: https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue.

R1NEJ0, 2017-12-04:
Do I open another thread?

Thanks!!

[files.zip](https://github.com/ccrisan/motioneyeos/files/1528862/files.zip)
[image]
[motion.log](https://github.com/ccrisan/motioneyeos/files/1528863/motion.log)

ccrisan, 2017-12-04:
Normally yes, but on motion's issue tracker, not here.

granoviter, 2017-12-15:
Hi. The resolution diffrence is normal?