motioneye-project/motioneyeos #1257: No motion mode with Rpi Zero

Skydive1, 2017-12-01:
Hi, 
I've motion version 4.0.1+git37b3595 and OS version motioneyeOS 20171003. All feature work perfectly but i've not motion mode. How can i fix it? 
Thanks

jasaw, 2017-12-03:
Can you please explain what you mean by not having motion mode? Attach screenshots if it helps.

Skydive1, 2017-12-04:
I've not motion mode in my setting

Skydive1, 2017-12-04:
[image]

portieay, 2017-12-06:
I am also having this same problem

ccrisan, 2017-12-07:
@Skydive1, @portieay have you enabled General Settings > Advanced Settings?

Skydive1, 2017-12-08:
@ccrisan yes i've enabled Advanced Setting

ccrisan, 2017-12-08:
Can I see a screenshot of your General Settings (opened)?

ccrisan, 2017-12-08:
Sorry, I meant Expert Settings.

Skydive1, 2017-12-08:
Wait i take screenshot