motioneye-project/motioneyeos #1258: Want to send a MQTT message on motion detection

DanielStefanK, 2017-12-01:
I am running MotionEye OS on my raspberry pi 3.

My question now is: Can I install mosquitto pub somehow to send a MQTT message or is there any other way?

Fairly new to this and I don't know how I can install this on the pie

jasaw, 2017-12-03:
Short answer is, no you can't easily install it on motionEyeOS. More info in the wiki: https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-i-install-some-package

If you want to do things outside the scope of motionEyeOS, you have 2 options:
1. Install motionEye on your favourite distro. Installation guide [here](https://github.com/ccrisan/motioneye/wiki/Installation). You can then use the package manager to install anything you like.
2. Clone, modify, and build your own motionEyeOS.

ejdudley, 2017-12-20:
Issue [#451](https://github.com/ccrisan/motioneyeos/issues/451) may be of interest, particularly at the end.