motioneye-project/motioneyeos #1262: homekit support

leonardpitzu, 2017-12-04:
can you add homekit support via nodejs + homebridge? motion detection and streaming through homekit would be amazing!

pauldogg, 2017-12-11:
Same for Home Assistant would be great!
Thanks!

SteveCohen, 2018-01-06:
I think stable homebridge plugins for this already exist. 
- Video can be streamed using [homebridge-camera-ffmpeg](https://github.com/KhaosT/homebridge-camera-ffmpeg/). 
- Motion detection info can be passed to homekit using motioneyeos' webhooks. A plugin like [homebridge-http-webhooks](https://github.com/benzman81/homebridge-http-webhooks) looks promising. 

Not as neat as a single 'all in one' plugin, but it works.

ccrisan, 2018-01-07:
There are no plans to integrate motionEye "natively" with homekit. Plugins could be a solution, as @SteveCohen indicated.

closed 2018-01-07 by ccrisan