motioneye-project/motioneyeos #1263: Issue with image loading on R B+ v1.2

kikkoskat, 2017-12-04:
If i load the image related to Raspb B+ version (1.2) nothing happen. The green led is fixed and no SO is loaded. What was wrong? The OS is this: motioneyeos-raspberrypi-20171003.img.gz

jasaw, 2017-12-04:
Did you extract the gz image before writing it to your SD card?
Have you tried writing the same image to the SD card again? Try another SD card if possible.
Double check that you really have a Pi 1.

kikkoskat, 2017-12-04:
tried everithing written by you...very strage, i am becoming crazy. i have
raspberry b+, v1.2 2014 (as printed on the board)

ccrisan, 2017-12-04:
How exactly do you write the image? Are you on windows, linux or osx?

kikkoskat, 2017-12-05:
I have windows and write the image using Win32 Disk Imager

ccrisan, 2017-12-05:
Then it beats me.

presiannedyalkov, 2017-12-05:
Hi, I just tried the last images and both the RPI Zero and RPI B 2 doesn't work for me.
For both, I use win32diskimager to write motioneyeos-raspberrypi-20171008.img.gz and motioneyeos-raspberrypi2-20171008.img.gz respectively. For the 0 I used the method with the wpa_supplicant.conf - it seems to work, because it shows on my router 'connected devices' list. I try to connect with the pi inputing the given ip:80 but it refuses the connection. Can't ssh either. Tried with the Pi B 2 connecting it to my tv with hdmi and lan cable. It can't get past the rainbow screen and as expected doesn't come up on the router's list as well. Also tried 2 different SD cards and installed and ran successfully a fresh Raspbian with Desktop on the B 2, just to be sure it's not due to faulty SD. I had no such problem in previous releases either with the Zero nor any of my other Pis.

kikkoskat, 2017-12-05:
Before mount images with win32.. have you unzip the file?

What is the last release in wich SO works on raspi 1?

presiannedyalkov, 2017-12-05:
win32diskimager allows only img files to be selected, so it is impossible to make a mistake there.

I tried a couple of releases on the B 2 spaning to 2016 without any luck. Finally after a while I found a working release for the pi 0 - motioneyeos-raspberrypi-20170329.img.

presiannedyalkov, 2017-12-05:
I figured out the problem with the 0 and it works now with the last release: when I was creating the wpa_supplicant.conf I copied the content from here: https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration with the tab spaces. I haven't done any other alterations but it works now and I can access the web interface.

Cheers.

kikkoskat, 2017-12-05:
And so no solution for raspi b+ version?

presiannedyalkov, 2017-12-05:
No, couldn't make it work

ccrisan, 2017-12-05:
@presiannedyalkov have you connected your B+ (running motioneyeos-raspberrypi-20171003.img.gz) to a monitor? If yes, what do you see?

presiannedyalkov, 2017-12-05:
Tried a new SD card on B+ (v1.2) with motioneyeos-raspberrypi-20171003 and it works.

ccrisan, 2017-12-06:
I believe we can close this issue now.

closed 2017-12-06 by ccrisan