motioneye-project/motioneyeos #1267: mmalcam_control_params don't work on motioneyeOS?

advancedNoob, 2017-12-08:
mmalcam_control_params are not working on motioneyeOS at all for example -vf -hf options used to work on motioneye installed on Raspbian. is this normal or am i doing something wrong ??

bmatic385, 2017-12-09:
That is because the latest MotionEyeOS 20171008 uses motion v4.0.1 from 2016. Motion option mmalcam_control_params is introduced in motion v4.1.

And now question for main developer, in which time period can we expect MotionEyeOS to be updated with motion v4.1 ?

advancedNoob, 2017-12-09:
@bmatic385 but 4.1 was released just a month ago and it was working even before that and it's mentioned in 4.0.1 release notes too.

bmatic385, 2017-12-09:
Yap You are right, my mistake.
May be helpful, from motion documentation:

Motion will ignore this option when the MMAL support is not included.

https://motion-project.github.io/motion_config.html#mmalcam_control_params

ccrisan, 2017-12-09:
MMAL support will be available in the next release. There's no ETA.

Alikhan500, 2018-01-05:
so i've updated to the Latest Pre-Release of MotionEyeOS and the motion_control_params are now working .. had to add mmal camera name too and after that it is working but the camera name option is not showing in the ui . All the options and camera effects are working. I have also tried the -md option to set resolution and frame rates but i don't know how the other options in motion for setting resolution and frame rates work undercover so i'm not sure about the behaviour

ccrisan, 2018-01-05:
@Alikhan500 thanks for getting back to us. Actually the MMAL settings are not covered by the motionEye UI and therefore all MMAL stuff appears in (and should be added to) the Extra Motion Options box.

Alikhan500, 2018-01-05:
yes i added the option there in the Options box. First i added the motion_control_params -vf -hf but it didn't work right away then in addition to that i added mmal_camera_name vc.ril.camera then the options started working and they are shown in the options box but the camera name option vanished from there . everything works now btw

ccrisan, 2018-02-06:
I think this issue can be closed, now that we have [20180101](https://github.com/ccrisan/motioneyeos/releases/tag/20180101) with full MMAL support.

closed 2018-02-06 by ccrisan