motioneye-project/motioneyeos #1273: MotionEyeOS crashes when streaming from Raspi Zero FNC to Pine64

GnomeTek, 2017-12-13:
I'm currently in process of setting up 2 raspiZero FNC cams to stream to a Pine64 board as my server. The RZeros are running fine and i can pick the streams off without issue in my web browser. The Pine64 is the same - no camera attached though. 

Once I add a Rzero cam to the Pine64, I see the active stream for ~30seconds, then i see my monitor attached to the Pine64 go back, and my browser portal hangs. Upon reboot I get a good monitor feed from the Pine64 until i hit "starting motioneye" once i see the "done" my monitor goes black and I am unable to connect via browser to the Pine64.

This last run I enabled various settings on the Pine64 - such as keep awake, and debug mode. And prior to adding the network cam nothing was reported in the logs. Now, with the network cam added I am unable to boot into CLI or the Web portal. 

Using a 32gB card in the Pine64. 

All network configs are good, and I should have enough overhead bandwidth for these streams (able to monitor from web portal, turn off web portal, and add cam to Pine64).

Any help would be appreciated!

GnomeTek, 2018-01-19:
Is this a code issue or do I have something mucked in the config? Can anyone help send me down a path with this?

ccrisan, 2018-01-19:
@GnomeTek there could be many causes for what you're experiencing. Please start with a plain setup, where your Pine64 has the two cameras added as Network Cameras and use the default configuration. If the problem persists, please try recovering the log files on your Pine64 (all of them) and attach them here. Please do **not** paste them here.