motioneye-project/motioneyeos #1274: Cannot access admin page

lodgemoorlad, 2017-12-14:
Hi, 
I have a working motiineye which is housed in a nest box . I can access the video using the ip address:8081,   but it's not allowing me access to the admin page, it's just saying the server is refusing access, very frustrating, but other than that it's worked well for over a year. Hope someone out there can help. Thanks in advance

ccrisan, 2017-12-14:
Use port 80 to access the admin page.

lodgemoorlad, 2017-12-14:
Hi Crisan
Thanks for a fast reply
I'm not very fluent in that area, do you know if theres a webpage you can
point me to that might help me/show me how to do this
Thankz

nolantomy, 2017-12-16:
address:8081 means you are accessing "address" on port 8081.
If try http://address that should automatically use port 80.
Or to be sure use http://address:80

Tom

lodgemoorlad, 2017-12-16:
Hi Tom
Thanks for the reply as I'm still struggling.
I'll try that in the morning as I'm out at the moment, but thanks for your
help, it's appreciated
Alex

billysyrett, 2018-06-15:
@lodgemoorlad - did you manage to resolve this?  I'm having the same issue.

jasaw, 2018-06-16:
@billysyrett Try pinging your Pi Zero. Try the troubleshooting steps here: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

billysyrett, 2018-06-16:
@jasaw I can ping the Pi Zero - the device connects to the WiFi fine and I can view the stream through 8081 but can't get to the admin interface.

machina-speculatrix, 2018-12-12:
I'm also having this issue. Got the system set up on a Pi3 B with fixed IP (10.0.60.51). I changed the port to 8081. Now when I go to http://10.0.60.51:8081 I get the video feed but no way to log into the admin page. Connections on http://10.0.60.51:80 are refused. I can SSH in but I'm not sure what I can do from there. Is there a config file I can edit to revert to port 80?

cismarine, 2019-03-22:
Ok, Looks like same problem here. 
I can ssh to it, but web interface does not work.
(accessing it via wifi on zero w)
Any ideas?

QSKONE, 2019-05-08:
> Ok, Looks like same problem here.
> I can ssh to it, but web interface does not work.
> (accessing it via wifi on zero w)
> Any ideas?
Same problem here

littlefatcat5963, 2019-09-08:
i have the same problem the setup is okaii if i access it with a wifi device but if i access it with the pc that is connect with a utp cable to the wifi router i cant access the device... what i want is to access it on my pc not on a laptop i have purchase a wifi usb adapter for my pc just to access it hope @ccrisan will address this issue and come-up with a solution

UnluckyForSome, 2020-04-16:
I have the same issue on both of my Raspberry Pi Zero's, I can ping them both, see them on the router as connected, but when I put their IP's into the browser address bar I get

"The connection has timed out

The server at 192.168.0.XX is taking too long to respond."

starbasessd, 2020-04-16:
If you are running the motionEyeOS the WebGUI is at
http://<ipaddress>
If you are running motionEye on Raspbian, the WebGUI is at
http://<ipaddress>:8765

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UnluckyForSome, 2020-04-16:
OK, so my Zero W is accepting the wpa_supplicant, and connecting to the internet, and is visible on the routers connected devices page with the same IP. However, when using a different device (like Fing on my iPhone) - it's not showing up as a device on the network, probably why I can't get to the login page on the browser of my PC by typing in the IP (192.168.0.97).

Can anyone let me know what's probably causing the issue?

[image]

starbasessd, 2020-04-16:
Why is your DNS set to 194.168.4.100? It's not a valid/active Internet IP
address, and not on your internal network
Looks like either the DHCP server isn't set up correctly, or you have
multiple internal networks, and the routes between them aren't set
correctly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UnluckyForSome, 2020-04-16:
I think it's the DNS address for my internet provider, Virgin Media? Is that wrong?

starbasessd, 2020-04-16:
Is your iphone connected to your home wireless?
If you use the browser (Opera?) and type in
http://192.168.0.97
Do you get anything?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

UnluckyForSome, 2020-04-16:
Thanks for your help starbasessd. I figured out what was wrong, I have a TP-Link powerline wireless extender, even though it should be set up properly and the SSID's set the same etc, I think it was somehow messing it up. I have unplugged it and I have access to my Pi's again! Weirdly I still have access after plugging the extender back in again too, so maybe it just needed a reset. Thanks again.

starbasessd, 2022-01-19:
Closing, no further comments

closed 2022-01-19 by starbasessd