motioneye-project/motioneyeos #1277: Randomly change video bitrate to very low [50~300 KiB] on Raspberry 3

granoviter, 2017-12-14:
Hi, We probe use the motionEyeOs on RPi3 but somtime the recorded video bitrate change randomly to very low bitrate. We use last image with base settings , raspberry Pi camera, and continuos recording (FullHD , or HD).

See the sample video on link:
https://drive.google.com/open?id=1aS9gcAL8Y3Qu8ETwrItFNrQH9ubFAPui

Mattncb, 2017-12-19:
I have the same issue, pi 2 with motion activated recording. Some videos recording perfectly with an average file size of 4mb but most are recording a video with tiny sub 100kb file sizes. Latest release and recording in omx mp4

ccrisan, 2017-12-20:
motionEye(OS) is not responsible for creating the video files. It's [motion](https://github.com/Motion-Project/motion) that takes care of creating those videos, indirectly using ffmpeg's libs and codecs. Please [report an issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue) to motion.

closed 2017-12-20 by ccrisan