motioneye-project/motioneyeos #1278: MotionEyeOS for Odroid C2: Enable using eMMC for OS, SD card for video storage

MarkusEakar, 2017-12-17:
Hello,

installing MotionEyeOS on SD card works pretty well with Odroid C2, but when installing MotionEyeOS on eMMC only eMMC can be selected and used for video storage.

Using both eMMC and SD card at the same time would be great as the „philosophy“ of seperating OS and video storage often is used in DVRs/VMS, so that disks/drives/etc. containing video storage can easily be ejected/changed.

So how can SD card be found/mounted, too, when MotionEyeOS is installed on eMMC?

ccrisan, 2017-12-18:
Hmm I'll see what I can do about that. I never tried it that way.

ccrisan, 2018-07-29:
@peoples0815 I haven't had time to take care of this issue yet. Pull requests are welcomed though :)