motioneye-project/motioneyeos #1279: admin password changes itself

bscuderi13, 2017-12-17:
Whenever I make a configuration change in motion eyeos that requires a reboot my password no longer works after the reboot. I'm unable to login... I just tried reinstalling a fresh image and resetting everything up. I changed the admin password and that worked fine I only made it as far as changing the timezone to az and it required a reboot. upon reboot I could again no longer login. :( Im running motioneyeos on an odroid xu4

ccrisan, 2017-12-17:
What OS version?

bscuderi13, 2017-12-17:
motionEyeOS 20171008

I just reinstalled again and this time I tried a password without special characters and it works

ccrisan, 2017-12-17:
It's either the well known issue #1116 that has to do with browser's cache or it's some special character that motionEye is not happy with. You could list here the special characters you're using (in some random order) if you want me to try investigating it.

bscuderi13, 2017-12-17:
just a $

bscuderi13, 2017-12-17:
oh maybe its the browsers cahce cause its doing it again without special characters

bscuderi13, 2017-12-17:
okay its resetting my password to ***** is there a fix or prevention for this yet?

ccrisan, 2017-12-17:
Yes, that's the browser cache issue. Clear it, refresh and it will stop doing that.

ccrisan, 2017-12-17:
In the meantime use five stars as password if you do not want to reinstall.

bscuderi13, 2017-12-17:
thnak you clearing refreshing worked and its back to my old password

closed 2017-12-18 by ccrisan

adrian0, 2018-05-31:
This is an infuriating bug. 
I have one pi zero w that I have been trying to set up for days now. Probably spent 8 hours going around in circles and continuously finding that I get locked out of my camera and unable to log in using the password that worked on previous attempts.
I can't find a pattern to this. It's definitely not the cache issue.

When i say it's infuriating, I'm now giving up and trialling other options - zoneminder etc. Arrgh.