motioneye-project/motioneyeos #1280: Action Buttons Not Appearing

s-moon, 2017-12-17:
Hi,

I just don't seem to be able to get the buttons to appear on my image.

I'm using:
```
motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008
```
on a Raspberry PI 3, using the downloaded package supplied via GitHub. My action buttons are stored in /data/etc and named lock_1, right_1 etc. (I've tried several). They are executable:

```
-rwxr-xr-x    1 root     root            29 Dec 17 21:13 lock_1*
```
My camera is definitely #1 and I only have one (as shown in the Video Device section).

I have also tried altering the resolution of the camera image, from 800x600 to 1280x960, without having any effect. Lastly, I have rebooted many times in case that was a factor and reloaded my image preview (stream) -- which is where I am expecting to see the buttons, though I don't seem them in the admin panel either.

Am I missing something? I've read the action button page but can't seem to get them to appear no matter what I try.

Thanks

ccrisan, 2017-12-17:
Can you attach a screenshot of your camera frame with overlaid stuff visible? Another one with the video device settings might also help.

s-moon, 2017-12-30:
[image]

This is a picture of my display through the camera. You can see that there aren't any buttons visible. Note: the black is because it is currently pointing at a wall, not because of any malfunction - the camera side of things works great. For info, you can just see the text at the bottom right showing it is working as expected.

[image]

These are my device settings. The camera is a Logitech C270 FYI in case that matters (https://www.amazon.co.uk/Logitech-C270-960-001063-HD-Webcam).

Sorry for the delay in responding @ccrisan - the Xmas effect hit me hard this year :-) 

Any help you could offer would be really appreciated - I love the package you've created; I just need to fix this shutdown issue.

ccrisan, 2017-12-30:
What happens if you click on the camera frame?

s-moon, 2017-12-30:
I see all the buttons. Groan! How did I miss that...thank you so much for your help; that resolves my issue.

closed 2017-12-30 by s-moon