motioneye-project/motioneyeos #1284: raspberry pi 3 will not load motion eye os at start

chefkarl, 2017-12-19:
I installed the motion eye os software that I downloaded from this site it was motioneyeos-raspberrypi3-20171008.img.gz. I copied it over using the win32 diskimager and all it does is go into raspian instead of loading motion eye. I tried the raspberry 2 download thinking I had the wrong version but that only loaded the rainbow screen.  I am going to try a older version of the sotware to see if that fixes the issue but I'm not hopefull.

chefkarl, 2017-12-19:
After going back to 20170710 I was able to get motion eye installed and I was able to upgrade to the latest version with no issues. I am trying 20171003 to see that one works.

ccrisan, 2017-12-20:
Given that you saw **raspbian** booting when you wrote 20171008 on your SD card, clearly the writing process did not complete. There are many reasons why that would happen. See [system won't boot](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot).

closed 2017-12-20 by ccrisan