motioneye-project/motioneyeos #1288: Raspberry Pi Zero W and No MMAL Camera Option

roger-zacharczyk, 2017-12-21:
When adding new camera there is no MMAL option.

Camera works as local camera (V4L2).

Thus there is no Fast Network Camera option.

ccrisan, 2017-12-21:
I'm sorry but I don't get what this is about. You listed 3 totally unrelated supported features of motionEyeOS. Could you be a bit clearer about what this issue is about?

roger-zacharczyk, 2017-12-21:
The goal is to enable FNC. I think this is available for MMAL camera only. But there is no MMAL camera option when adding new camera.

ccrisan, 2017-12-21:
You don't add FNC to motionEye. In fact FNC has almost nothing to do with motionEye. When in FNC mode, motionEye won't allow you to add or remove cameras - it will statically display the MJPEG stream of the preconfigured FNC.

One more thing: you enable the FNC from Expert Settings - you don't add it. I was under the impression that this was clear from the beginning.

roger-zacharczyk, 2017-12-22:
Ok, let me describe what I did.

1. I wrote the raspberrypi image on sd card
2. Started my RPI Zero W with this sd card
3. connected to the ip:80 in web browser
4. there is no FNC option in the expert settings

ccrisan, 2017-12-22:
Now I get it. What OS version is this? Can I see a screenshot of your expert settings?

jasaw, 2018-06-05:
@roger-zacharczyk Did you manage to get the FNC option? If not, try running the latest firmware.

jnovack, 2018-06-19:
While the motion version and the motioneye version are the same on disk, the websites are different.

Raspberry Pi 2:
[image]

Raspberry Pi Zero:
[image]

I'm pretty damn sure I used the same image.  I can also confirm there's no "Fast Network Camera".  What's going on?

ccrisan, 2018-06-19:
You forgot to enable Advanced Settings.

closed 2018-06-19 by ccrisan

jnovack, 2018-06-19:
Today, it was me. 

[image]