motioneye-project/motioneyeos #1290: Disabling camera LED on Rpi Zero

jwc866, 2017-12-23:
Hello:
I am trying to turn off the red LED light on my camera module on a Raspberry Pi Zero.  I followed the instructions from the wiki for the Zero:
https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs
It says "dtparam=act_led_activelow=on", this should be ON and not OFF?  I have tried it both ways, set to ON and OFF, however the light is still on.  I have confirmed the lines are in the /boot/config.txt file and correct, after a reboot.  Any ideas?
I appreciate the help.  Thank you.

kefabean, 2017-12-24:
Why not try adding this to /boot/config.txt

disable_camera_led=1

Your command appears to be attempting to disable the activity LED on the Raspberry Pi board.

jwc866, 2017-12-25:
Yes, that worked, thank you!  I was thinking the article in the wiki was for the camera but yes it appears it is for the Pi board.

jasaw, 2018-01-14:
Issue appears to be resolved. Closing.
Just FYI, you can disable camera LED from motionEye web interface, under the expert settings.

closed 2018-01-14 by jasaw

giuse320, 2018-12-18:
> Why not try adding this to /boot/config.txt
> 
> disable_camera_led=1
> 
> Your command appears to be attempting to disable the activity LED on the Raspberry Pi board.

Thanks ,it works !!!