motioneye-project/motioneyeos #1291: Action Button says An Error Occurred Refreshing Is Recommended

huvarda, 2017-12-24:
I have an action button script called alarm_on_1 and the function is `aplay /home/ftp/sdcard/Siren.wav
` The button shows up on the camera but when I press it it says "An error occurred. Refreshing is recommended"
Log Files:
[boot.log](https://github.com/ccrisan/motioneyeos/files/1584658/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1584659/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1584661/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/1584662/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1584660/motioneye.log)

ccrisan, 2018-02-05:
Your log is full of `URLError: <urlopen error [Errno -3] Temporary failure in name resolution>`. Please make sure you have proper internet access. More precisely, that you have a working DNS server configured.

pmatias, 2019-05-05:
I'm getting the same error, but in my case I see:

```
2019-05-01 16:07:05: [motioneye]    ERROR: 500 POST /config/0/set/?_=[redacted]&_username=[redacted]&_signature=[REDACTED](192.168.[redacted]) 56.28ms
2019-05-01 16:07:16: [motioneye]     INFO: rebooting
2019-05-01 16:07:16: [motioneye]     INFO: executing "/sbin/reboot"
2019-05-01 16:07:18: [motioneye]     INFO: interrupt signal received, shutting down...
2019-05-01 16:07:18: [motioneye]     INFO: server stopped
2019-05-01 16:07:18: [motioneye]     INFO: tasks stopped
2019-05-01 16:07:18: [motioneye]     INFO: motion stopped
2019-05-01 16:07:18: [motioneye]     INFO: smb mounts stopped
2019-05-01 16:07:18: [motioneye]     INFO: bye!
2019-05-01 16:07:42: [motioneye]     INFO: hello! this is motionEye server 0.40
2019-05-01 16:07:45: [motioneye]     INFO: cleanup started
2019-05-01 16:07:45: [motioneye]     INFO: wsswitch started
2019-05-01 16:07:45: [motioneye]     INFO: tasks started
2019-05-01 16:07:45: [motioneye]     INFO: mjpg client garbage collector started
2019-05-01 16:07:45: [motioneye]     INFO: smb mounts started
2019-05-01 16:07:45: [motioneye]     INFO: server started
2019-05-01 16:08:59: [motioneye]    ERROR: 'prereleases'
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1499, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1700, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 290, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 522, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 371, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 449, in set_camera
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 242, in set_main
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 2072, in _set_additional_config
  File "/usr/lib/python2.7/site-packages/motioneye/extractl.py", line 188, in _set_os_settings
KeyError: 'prereleases'
```

ccrisan, 2019-05-19:
@pmatias your error has nothing to do with the OP's. In your case, make sure you run the latest version of motionEye(OS) and that your have cleared your browser's cache.

closed 2019-05-19 by ccrisan

pmatias, 2019-05-20:
Hi @ccrisan, thanks for your answer.

I was running the latest dev version, and I recently updated to dev20190511. Tried different browsers, different machines (that never accessed the web ui before). Same issues.