motioneye-project/motioneyeos #1294: Changes Date / Time writing Images

Dobi29, 2017-12-25:
Hi,
i have the problem with writing images by a motion. Somtimes the date / time is correct an two hours later he uses the folder two days in the past and the wrong time....two hours later everything is ok.....
Can someone help me ?
Thx
Klaus

ccrisan, 2017-12-27:
Please [attach your log files](https://github.com/ccrisan/motioneyeos/wiki/Reporting-An-Issue#attach-log-files) here (don't send them by replying to the email, please).

Dobi29, 2017-12-27:
log files
[boot.log](https://github.com/ccrisan/motioneyeos/files/1589478/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1589479/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/1589480/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1589481/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1589482/motioneye.log)

ccrisan, 2017-12-28:
motionEyeOS 20160919? Isn't this a bit old? Can you try the latest stable version and see if the problem persists?

Dobi29, 2017-12-28:
what is the lastest stable version ? can i use the update button to get this version ?

ccrisan, 2017-12-28:
I wouldn't recommend that, given the large number of versions in between.

Dobi29, 2017-12-28:
is it possible to export the settings und than import to the new version ?

ccrisan, 2017-12-28:
Again, not recommended.

Dobi29, 2018-04-15:
i have the now problem with this version:
[image]
the date changes...here are my log files
[messages.log](https://github.com/ccrisan/motioneyeos/files/1912158/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1912159/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1912160/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/1912161/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1912162/dmesg.log)

Dobi29, 2018-04-20:
...no help ? no ideas ?

jasaw, 2018-06-05:
@Dobi29 A few questions:
- Are you connecting via wifi or Ethernet?
- Which time sync method did you choose? NTP or HTTP?

Can you please describe your problem clearly? Are you getting time drift problem or getting year 1970 timestamp on your media files? When does it happen? Does it recover after a while?

Dobi29, 2018-06-06:
Hi
-connecting via Ethernet to the rooter with surfstick. I'm testing at the moment if the internet connection with the surfstick is stable. My vpn conection to the rooter works fine.
- Date Method --> HTTP
- new installation last sunday --> 20180401 --> same problem date 1970, then correct date, today wrong time and two days in the past

jasaw, 2018-06-07:
That's really odd. Even if it didn't manage to time sync in 5 days due to network issues, I wouldn't expect time to drift by 48 hours within 5 days. Which HTTP time server did you use?
Have you tried NTP time sync method?

Dobi29, 2018-06-08:
Yeterday i changed the HTTP time server to "time.google.com" it works fine at the moment.

Do you have a solution for double files in the upload

[image]

jasaw, 2018-06-08:
@Dobi29 Are they are the same photo or different photos taken at sub-second interval? Can you run a `diff` on the 2 files?

ccrisan, 2018-06-08:
@Dobi29 add -%q to the filename pattern.