motioneye-project/motioneyeos #1295: Delete files afters x days not working

joehas, 2017-12-27:
Hi, I've never been able to get the delete movies older than x days feature working. My movies are stored on a NAS and I have selected that they are deleted after 15 days. Which logs can I look at to see what's stopping this from working?

Many thanks Joe

Running on a Raspberry Pi 3
motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008

[image]

ccrisan, 2017-12-28:
Please attach your motioneye.log file.

Chiny91, 2017-12-28:
I had always assumed that the "delete older than..." feature applied only to movies/pictures on the motionEyeOS device itself, and did not extend to other storage systems like a NAS.   Certainly my (still) pictures don't get deleted after the config lifetime; but maybe that is because they are rsync'd to NAS, not GUI'd.

ccrisan, 2017-12-28:
The NAS is actually mounted locally, so it qualifies as a local folder. Therefore the deletion mechanism should work just as well. If it doesn't, it's a bug :)

joehas, 2017-12-29:
Hi @ccrisan ,

Please find log attached below.

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1592467/motioneye.log)

ccrisan, 2017-12-29:
@joehas can you please turn on debugging (in Expert Settings) and leave the system running for two days? Then please send me the motioneye log file corresponding to these two days.

On a side note, from what I can see in your current log, you have some connectivity issues with the FTP upload service you configured. Also you seem to store files on a NAS and also upload them to an FTP server. While this is fine in theory, it doesn't make too much sense as a use case and the network I/O may pose problems.

joehas, 2018-01-05:
Hi @ccrisan 
find log attached 
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/1606756/motioneye.1.log)

I'll take a look at only storing the files once, I seem to remember I couldn't get it working just using FTP, worth another shot though.

Joe

ccrisan, 2018-01-07:
@joehas the log looks alright. The cleanup procedure appears to run regularly and to complete without error. The only explanation I can come up with right now is that the SMB mount somehow reports wrong date/time. I'll try to dig deeper into this and see if I can reproduce it.

mseriesmseries, 2018-02-06:
This functionality doesn't seem to work for me either. I have enabled logging as instructed above - what do we need to look for in the log file ?. I am saving the movies on the SD card.

ccrisan, 2018-02-08:
@mseriesmseries you could manually ssh into your running system and see what creation dates it reports on the media files. The same applies to @joehas, only that the files should be under the mounted SMB folder.

If the dates are indeed correct and the system date is also correct, then you should look through the log file for any cleanup errors.

Lines containing "cleanup", "removing file" and "failed to remove" could indicate more details on the problem.

mseriesmseries, 2018-02-08:
Thanks @ccrisan . It does actually seem to work, I didn't realise how frequently it performs it's checks to determine if anything can be deleted so I have files that were well over 1 day old. Is it possible to make it check more frequently say every hour ?

ccrisan, 2018-02-09:
@mseriesmseries see https://github.com/ccrisan/motioneyeos/wiki/motioneye.conf. You're particularly interested in [cleanup_interval](https://github.com/ccrisan/motioneye/wiki/Configuration-File#cleanup_interval).

closed 2018-02-09 by ccrisan

Moonbase59, 2018-03-18:
@ccrisan: I also have this problem: Taking stills (to be deleted after 2 days) doesn’t remove the old ones images.

Raspberry Pi Zero W
motionEye Version | 0.38.1
Motion Version | 4.1.1
OS Version | motionEyeOS 20180224

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1822583/motioneye.log)